supermerill/SuperSlicer #3771: Wild external perimeters

Xeroxxx, 2023-04-03:
### What happened?

Currently improving my printer with parts. However I get strange external perimeters that are actually not present within the model. Its easy to remove with pliers but I do not understand how this happens, as it doesnt in prusaslicer.

[image]

### Project file & How to reproduce

[Tap_Magnet_Left_r2.zip](https://github.com/supermerill/SuperSlicer/files/11136863/Tap_Magnet_Left_r2.zip)

### Version

Version 2.5.59

### Operating system

Windows 10

### Printer model

Voron 2.4r2

DVSVIDEO, 2023-04-05:
The same was observed with the calibration cube on the finished print. They are in the X and Y recesses.
In addition, on one occasion there was a strange movement: the head went out to the X and Y max without extruding, then went back in and continued printing.
So many BUGs have accumulated in the SuperSlicer that I am suspending its use.

DVSVIDEO, 2023-04-05:
https://github.com/supermerill/SuperSlicer/issues/3769

Xeroxxx, 2023-04-05:
Duplicate

closed 2023-04-05 by Xeroxxx