supermerill/SuperSlicer #3774: Paint-on Fuzzy Skin

LouisWatermeyer, 2023-04-03:
**Is your feature request related to a problem? Please describe.**
Fuzzy Skin is a great feature, but the existing options (Outside walls, External walls, App perimeters) are not very precise.
This is a problem if you have a functional part that connects to something else but you want it to be grippy/fuzzy on a specific side.
It is technically possible to achieve this with object modifiers, but that can be some pain and I think it would be a handy feature for people who like fuzzy skin.

**Describe the solution you'd like**
A potential solution would be to provide "Paint-on Fuzzy Skin" similar to the "Paint on Supports" feature. It would be cool if we could add "Fuzzy Skin Enforcers" and "-Blockers" this way.

**Describe how it would work**
It would work just like the "Paint-on Supports" feature.

**Describe alternatives you've considered**
Technically modifiers work, but paint-on would be nicer.

**Additional context**
I always wanted to contribute to this project and already tried looking into the necessary changes. If there are some people interested in this feature, I would invest time into that and create a PR if I get it working.

Rough idea visualized:
[image]

[image]

DVSVIDEO, 2023-04-05:
Yes, I am interested. I've reported Prusa SLicer bugs and requests (and many others)
https://github.com/prusa3d/PrusaSlicer/issues/9799
https://github.com/prusa3d/PrusaSlicer/issues/9353
https://github.com/prusa3d/PrusaSlicer/issues/9787
https://github.com/prusa3d/PrusaSlicer/issues/10131
https://github.com/prusa3d/PrusaSlicer/issues/8357

LouisWatermeyer, 2023-04-05:
#https://github.com/prusa3d/PrusaSlicer/issues/6988

Found this in my research. He uses special STLs as modifiers, but this seems very time-consuming.

supermerill, 2024-02-01:
At this point, It's not hader to create a "modifier paint" thing.

Michl59, 2024-08-18:
+1 for this feature :)