supermerill/SuperSlicer #3777: exported file unrecognized by printer.

userumberlla, 2023-04-04:
### What happened?

After I sliced the model and upload to the printer, (I'm using mainsail), the file would be displayed with no information at all. I tried to print the file, but it will automatically cause the printer to froze at a "paused during print job" state, I tried to hit both continue print buttom (green triangle) and the cancle print buttom, both don't do anything any more and I have to restart the entire printer.

Same file work just fine on 2.4 version as I tested.
I manage to fix it for a few times when I switched the unites from metric to inches back and forth, but its not a stable fix.
[image]

### Project file & How to reproduce

[test.zip](https://github.com/supermerill/SuperSlicer/files/11143996/test.zip)
My guess the issue is on the "switch to inches" option
1. Open a inch based model and close it
2. Open a previous 3mf file with a metric unit
3. Now in the slicer the "inches" icon is till on, but the actual number is still in mm
4. Upload file to printer
5. printer cannot recognized the file.

### Version

2.5.59.2

### Operating system

windows 11

### Printer model

voron 2.4 afterburner 0.6 nozel 0.2mm

gaifeng8864, 2023-04-10:
> ### What happened?
> After I sliced the model and upload to the printer, (I'm using mainsail), the file would be displayed with no information at all. I tried to print the file, but it will [...]

Have you checked the Gcode file? Are there garbled characters in it? I used the 2.5.59.2 version for the first time today and encountered similar problems. However, the reason for the problem I encountered was that the newly added "Klipper-flavor EXCLUDE_OBJECT" could not correctly recognize the Chinese name, resulting in garbled characters.
https://github.com/supermerill/SuperSlicer/issues/3790

gaifeng8864, 2023-04-13:
> I think its the Chinese name issue too, the stl before it is import to slicer probably cannot be chinese names
> […](#)

Is the problem with your file also because of the Chinese name?

userumberlla, 2023-04-14:
yes I've tested it yesterday, when importing the stl file in to the slicer,
the file cannot be in chinese characters, if it is, the file would be
"corrupted" as even the name of the object got changed in the slicer into
english, it would still not be recognized by klipper.(Basically the entire
3mf file would be unuseable even opened it again with 2.4 version)  However
if we changed the file name into english before importing, it would not
have any issue.