supermerill/SuperSlicer #3783: Arachne doesn't generate top surfaces properly

theodiem, 2023-04-06:
### What happened?

When using arachne, top surface, except for the tallest one, doesn't get generated with the same "top surface settings", for example single perimeter and flow/speed settings.

First screenshot with "classic" selected, second screen with only changing to arachne
https://imgur.com/a/RwxSJRz

### Project file & How to reproduce

(For whatever reason I cannot upload the files)
But it happens with any file I use, if I keep "classic", top surfaces are as expected.

### Version

2.5.59.2

### Operating system

windows 11

### Printer model

voron 2.4 with afterburner/galileo

gaifeng8864, 2023-04-10:
> ### What happened?
> When using arachne, top surface, except for the tallest one, doesn't get generated with the same "top surface settings", for example single perimeter and flow/speed settings.
> 
> First screenshot with "classic" selected [image]
> 
> second screen with only changing to arachne [image]
> 
> ### Project file & How to reproduce
> But it happens with any file I use, if I keep "classic", top surfaces are as expected.
> 
> ### Version
> 2.5.59.2
> 
> ### Operating system
> windows 11
> 
> ### Printer model
> voron 2.4 with afterburner/galileo

I have read the instructions, the arachne mode is like this, it is not a BUG. But I also want the single outline top in arachne mode, it's very beautiful.

theodiem, 2023-04-11:
> 
> I have read the instructions, the arachne mode is like this, it is not a BUG. But I also want the single outline top in arachne mode, it's very beautiful.

Can you provide a link to said instructions?

I find a bit odd because the tallest surface does take the "only one perimeter" in consideration. If it is the intended behavior I find at least a bit confusing since the affected option is neither "grayed out" or something that tells it works "only in a given condition".

closed 2023-04-11 by theodiem