supermerill/SuperSlicer #3785: Overhang threshold transforming curves into straight lines

theodiem, 2023-04-06:
### What happened?

After migrating to 2.5 beta, I've noticed that overhang lines are being converted to straight connections.
I know it makes sense in a bridge which is kinda very very difficult to make corners in a unsupported line .. but overhangs are not the case. I would like to apply overhang speeds and fan but not treat those lines as "bridges".
[https://imgur.com/a/JOwgkEj](sccreenshot)

(No idea why I cannot attach images, it always report an issue that can't process the file).

### Project file & How to reproduce

1. Draw a cilinder
2. make a tapered hole inside it (so you create overhangs on the inside)
3. set your overhand threshold to lower the angle (but keep the bridge flow threshold higher than that, so the lines aren't marked as bridge)
4. watch and cry as the slicer makes 4 straight lines in your perfect overhung circle.
5.

### Version

2.5.59.2

### Operating system

windows 11

### Printer model

voron 2.4 with afterburner/galileo

Kerman-ap, 2023-04-24:
having the same issue here

blandown, 2023-05-20:
Same here.
When unchecking the "threshold for" option in "Overhangs", the problem disappears

[image]
[image]
[image]

abrasius, 2023-06-17:
> Got this interesting slicing result with runaway external perimeter: [image]
> 
> ..And got it to behave by adding curve smoothing of 0.1mm: [image]

theodiem, 2023-09-04:
Unfortunately that solution doesn't work on my models :( .. would have loved if it did. Makes a bit difficult to use overhang speed modifier on 2.5 :(

VanessaE, 2023-10-02:
Another thing I've found that sometimes helps to mitigate this bug is simply changing the (base) layer height.  For example if it's 0.2 mm, lower it to 0.17 mm or so, and the problem goes away, at least at that moment.

j-snijder, 2023-10-02:
Not working for me, as moving the object away from mid bed isn't either.

It could be something with profiles because when I loaded a profile from an older cura the problem also happened on files that sliced okay before.

Anyway I reverted to 5.3.1 and it looks like its okay, for now.

Thanks for a good slicer anyway 😁

Murrdo, 2024-01-01:
[image]

The bug remained in the latest version, I checked it against the version taken from here - 
https://github.com/supermerill/SuperSlicer/releases/tag/2.5.59.6
And here - 
https://github.com/supermerill/SuperSlicer/actions/runs/7372223946
[model.zip](https://github.com/supermerill/SuperSlicer/files/13802584/model.zip)
l/SuperSlicer/actions/runs/7372223946

Murrdo, 2024-01-05:
I removed all print and printer profiles, the defect was reduced by 95%
(when cutting arachne and the threshold for overhangs is set)
Now artifacts appear if you change the extrusion width (for perimeters) other than the default one.
For example:
[image]
Gives:
[image]

But after a couple of saves and changes in the wall thickness, the defect disappeared...
I don’t understand what exactly causes the artifacts to appear, but now it’s clearly related to the extrusion width settings

closed 2024-02-11 by supermerill