supermerill/SuperSlicer #3786: app crashes after opening

vanrri, 2023-04-08:
### What happened?

after popup screen the following message appears:

"Unhandled unknown exception; terminating the application." 

then:

"Internal error: Calling AppConfig::save() from a worker thread!"

[image]

[image]

### Project file & How to reproduce

[Untitled.zip](https://github.com/supermerill/SuperSlicer/files/11182197/Untitled.zip)

### Version

2.4.58.5

### Operating system

windows 10

### Printer model

voron 2.4 300mm SB

ChristianReil, 2023-06-01:
Same here

[image]

daehahn, 2023-06-02:
Same issue on PrusaSlicer, refer below.

https://forum.prusa3d.com/forum/prusaslicer/prusa-slicer-unhandled-unknown-exception-terminating-the-application-internal-error-calling-appconfigsave-from-a-worker-thread/