supermerill/SuperSlicer #3788: External perimeters partly detaches

Neelix96, 2023-04-09:
### What happened?

That problem happened occasionally, but I finally got the time to at least try to file a good report.
I sliced my object with SuperSlicer_2.5.59.2_2023-02-24 nightly build and this happended:

[image]

At first, it looks like stringing, but in the slicer this is also showing:

[image]

It looks like, the External perimeters are sometimes detaching from the object and simply bridge to the other side. I couldn't find a setting to remove these "strings" so I sadly have no ideas, where the prolem might be.

Target Geometry:
[image]

### Project file & How to reproduce

.zip contains 3mf and gcode of the project

[superslicer_Bug.zip](https://github.com/supermerill/SuperSlicer/files/11184947/superslicer_Bug.zip)

### Version

2.5.59.2_2023-02-24

### Operating system

Windows 10

### Printer model

voron 2.4 with afterburner

lightoze, 2023-04-13:
I just hit the same bug. Opening the same project in 2.4.58.5 works fine.

mrulke, 2023-04-14:
same issue here

JanB97, 2023-04-14:
Known issue with that version: #3681