supermerill/SuperSlicer #3790: Klipper-flavor EXCLUDE_OBJECT does not recognize OBJECT with Simplified Chinese name

gaifeng8864, 2023-04-10:
### What happened?

Klipper-flavor EXCLUDE_OBJECT does not recognize OBJECT with Simplified Chinese name.
After I import an object with a Simplified Chinese name for slicing, superslicer2.5.59.2 generates Gcode, but it doesn't work when I upload to klipper for printing. After my investigation, it was confirmed that the newly added "Klipper-flavor EXCLUDE_OBJECT" could not recognize the Chinese name. After I commented out the sentence about "Klipper-flavor EXCLUDE_OBJECT" in the Gcode file, it returned to normal. In the previous version Although the "mark object" function is also enabled, there is no such problem. It seems that the newly added "Klipper-flavor EXCLUDE_OBJECT" caused this error.
[image]

### Project file & How to reproduce

[Klipper-flavor EXCLUDE_OBJECT.zip](https://github.com/supermerill/SuperSlicer/files/11190386/Klipper-flavor.EXCLUDE_OBJECT.zip)
1.Use superslicer2.5.59.2 to slice 新版测试打印效果盒子.3mf
2.Upload Gcode to klipper for printing.
3.You'll notice that printing doesn't happen automatically,And the thumbnail will not be displayed on the fluidd interface.
4.At this time, open the Gcode file, and there is a garbled code after "EXCLUDE_OBJECT_DEFINE NAME=". If you comment out this line of code, the Gcode file will return to normal, but EXCLUDE_OBJECT will not be available.

### Version

2.5.59.2

### Operating system

Windows10

### Printer model

Assemble it yourself, using klipper.

ix-prism, 2023-06-29:
I uploaded same error,  seems the filename can only be UTF-8

gaifeng8864, 2023-07-01:
> I uploaded same error, seems the filename can only be UTF-8

This should be. Unfortunately, the author's updates have stalled.