supermerill/SuperSlicer #3792: Top Layer Quality Question

xenon2008, 2023-04-12:
### Version
SuperSlicer_2.4.58.5

### Operating system type + version
Windows 10 Pro x64 

### Behavior
If i printed a File, the Top Layer has some visible "lines" there.
I tried many things in the last Days.
From other Infill Types, other Speeds, other Temps, other Material and other Slicer Settings....
Today i printed the same File with the same settings (the most important settings such as retract, speed, temperature and so on) with PrusaSlicer, and the TopLayer is really nice.

I also tried other SuperSlicer Profiles, but i didnt found a solution.

Printed with SuperSlicer:

[image]

And the same printed in PrusaSlicer:

[image]

I also tried same line Withs in SuperSlicer like PrusaSlicer ueses it...

Sorry i hope everybody could unterstand me & know what i mean, because my English Skills are not very Good :(

Here is one Project File for Example:

#### Project File (.3MF) where problem occurs
[Koerper10_SUSI TestSettings_Upload.zip](https://github.com/supermerill/SuperSlicer/files/11211721/Koerper10_SUSI.TestSettings_Upload.zip)

Could somebody help or say me what is wrong here?

xenon2008, 2023-04-12:
[Prusa Example.zip](https://github.com/supermerill/SuperSlicer/files/11213199/Prusa.Example.zip)

I created a new one, because i deleted the other already.

neophyl, 2023-04-12:
Ok one really big difference that struck me immediately is in your Printer Settings.  Both the SuSi and PS projects have the printer configured to use Relative Extrusion.   However your Prusa settings have 
;BEFORE_LAYER_CHANGE
;[layer_z]
G92 E0

while that is missing from the SuSi profile, its blank in that field.  

The reason its there in the prusa profile is they (as in Prusa) no longer allow you to slice without it being in there if Relative extrusion is enabled.  That's because they realised that without resetting E regularly (basically every layer) it can lead to over extrusion from rounding errors.  SuSi doesn't have that check in at the moment but its a pretty fundamental difference.

I'm not saying that this is the cause, what I am saying is that it 'could' be.

Other notable differences are the bridging extrusions on top of the infill.  The PS project has thick bridges enabled (the old default) which means they are double thick and also have a width of about 0.4mm same as the nozzle.  The same layer in Susi has a bridge flow ratio of 70% so the extrusion width is roughly 0.28-0.29mm (Layer 12).  The lines are more densely packed.  If you have any over extrusion at all then its going to build up there during that layer.  

You also have 1 perimeter on top layers turned on.  While that doesn't 'appear' to be doing anything untoward it has a long history of screwing things up.  Whenever I see anything weird going on its the first thing that gets turned off :)  Just in case.

That top layer is being printed at around 100mm/s in PS and about 50mm/s in SuSi.  

PS is set to lift Z between 0.2 and 500mm and SuSi , well isn't.  

Theres probably others but that should be enough to go on with.

northcoastam, 2023-04-12:
Take a look at [monotonic infill](https://blog.prusa3d.com/everything-you-need-to-know-about-infills_43579/#:~:text=The%20monotonic%20infill%20pattern%20fills,never%20in%20the%20opposite%20direction.).  I think thats the difference.  If you look at the visible lines in the SS print you will notice that they are tangent to either holes or raised features on the surface.

neophyl, 2023-04-13:
If you had taken the time to look at the attached projects you would have seen that both are already set to montonic top layers.

northcoastam, 2023-04-13:
Ah, sorry for the noise.

xenon2008, 2023-04-13:
> Ok, ein wirklich großer Unterschied, der mir sofort aufgefallen ist, sind Ihre Druckereinstellungen. Sowohl im SuSi- als auch im PS-Projekt ist der Drucker für die Verwendung von Relative Extrusion konfiguriert. Ihre Prusa-Einstellungen haben jedoch ;BEFORE_LAYER_CHANGE ;[layer_z] G92 E0
> 
> während das im SuSi-Profil fehlt, ist es leer in diesem Feld.
> 
> Der Grund dafür, dass es im Prusa-Profil vorhanden ist, ist, dass sie (wie in Prusa) Ihnen nicht mehr erlauben, zu schneiden, ohne dass es dort drin ist, wenn die relative Extrusion aktiviert ist. Das liegt daran, dass sie erkannt haben, dass es ohne regelmäßiges Zurücksetzen von E (im Grunde jede Schicht) zu einer Überextrusion durch Rundungsfehler kommen kann. SuSi hat diesen Check-in im Moment nicht, aber es ist ein ziemlich grundlegender Unterschied.
> 
> Ich sage nicht, dass dies die Ursache ist, was ich sage, ist, dass es sein "könnte".
> 
> Andere bemerkenswerte Unterschiede sind die überbrückenden Extrusionen auf der Füllung. Das PS-Projekt hat dicke Brücken aktiviert (der alte Standard), was bedeutet, dass sie doppelt dick sind und auch eine Breite von etwa 0,4 mm wie die Düse haben. Dieselbe Schicht in Susi hat ein Brückenflussverhältnis von 70 %, sodass die Extrusionsbreite ungefähr 0,28–0,29 mm beträgt (Schicht 12). Die Linien sind dichter gepackt. Wenn Sie überhaupt eine Überextrusion haben, wird sie sich dort während dieser Schicht aufbauen.
> 
> Sie haben auch 1 Umkreis auf obersten Ebenen aktiviert. Das scheint zwar nichts Ungewöhnliches zu tun, hat aber eine lange Geschichte, Dinge zu vermasseln. Immer wenn ich etwas Seltsames sehe, wird es als erstes ausgeschaltet :) Nur für den Fall.
> 
> Diese Deckschicht wird mit etwa 100 mm/s bei PS und etwa 50 mm/s bei SuSi gedruckt.
> 
> PS ist so eingestellt, dass es Z zwischen 0,2 und 500 mm anhebt, und SuSi ist es nicht.
> 
> Es gibt wahrscheinlich noch andere, aber das sollte reichen, um weiterzumachen.

Good Morning,
Yeah i know that some things are other in the Prus Project, this is because i deleted the orignal one already and created "fast" this project with the most important things.

The hint about the Bridges is a Good Point i will test this!
also the one Perimeter Setting i would test (when disabled), but if I remember correctly i enabled this in the PS Test print with the "orignal" (deleted) PS File.

Regards the Top Layer Speeds i tried "everything" already, From very slow to very fast, so thats not the problem ;)

Kindly Regards

neophyl, 2023-04-13:
Prusa Slicer does not have the 1 perimeter on top layers setting btw.  Its one of those SuSi specific settings.  The main one I would test first is the resetting of E in the between layers gcode field.

xenon2008, 2023-04-13:
> Prusa Slicer does not have the 1 perimeter on top layers setting btw. Its one of those SuSi specific settings. The main one I would test first is the resetting of E in the between layers gcode field.

Good idea!
You mean with Susi right?

neophyl, 2023-04-13:
Yes, I would add the G92 E0 to the Susi printer profile and test.

Pavulon87, 2023-04-13:
I don't think that will change anything.
In your gcode file you should already have:
; custom gcode end: layer_gcode
G92 E0 ; reset extruder position to flush any extruder axis rounding

neophyl, 2023-04-13:
Is no one actually examining the projects that xenon attached ?  
The point is that the Printer Profile in the SuSi project does NOT have the G92 E0 to clear the rounding errors but the PS project Printer profile DOES.  As I pointed out that may possibly be the cause.  Only testing by Xenon will determine that.

Pavulon87, 2023-04-13:
> Is no one actually examining the projects that xenon attached ? The point is that the Printer Profile in the SuSi project does NOT have the G92 E0 to clear the rounding errors but the PS project [...]

Because with Prusa you are forced to manually insert G92 E0. SuSi inserts is automatically

[image]

Pavulon87, 2023-04-13:
I'm not sure when it was added (certainly before 2.3.57), but the fact that you have to add G92 E0 manually I found out recently when I was testing a new version of Prusa. I've never had to do this before ;)

I'm sure these lines are a matter of different path planning.
[image]

Pavulon87, 2023-04-13:
This looks even worse in 2.5.59 (layer 16): 
[image]

xenon2008, 2023-04-13:
I'll try all that, I think, at the weekend because it's very difficult for me during the weekday. But thanks anyway to everyone who helps me here ;)

xenon2008, 2023-04-16:
Good Evening everybody,

I think i found the Problem!

It was because of the bridge density as it looks.
I reduced the default value from 100% to 50% and the result got a lot better.
[image]

Unfortunately, my PETG green is now off, so I did another test with black, but that's very difficult to see & photograph.

If I can, I will do more test prints in the next few days, but I think I know what the problem is now =)
Many thanks to all of you for your tips and your help with troubleshooting!

Best regards

Dx-Dx-Dx, 2023-04-18:
> If you had taken the time to look at the attached projects you would have seen that both are already set to montonic top layers.

Montonic connected does not work ( not montonic but rectilinear ) in last few versions. Use monotonic not connected
[image]

xenon2008, 2023-04-19:
This is a very useful hit. 

I didnt know that! - So Thanks ;)