supermerill/SuperSlicer #3797: Arachne - Gaps in Perimeter walls and Infill?

EmreDogann, 2023-04-16:
### What happened?

I'm trying to slice a spacer but I'm been having problems with gaps in the perimeter walls and infill when using the Arachne engine (red circles). These gaps are present in every other model I've sliced with Arachne and happen at points of small features.

Also, It is a little hard to see from the screenshot, but the side wall is supposed to be flat/straight but there are slight bulges from the places where the perimeter widths are increased (black arrows). These bulges seem slight but noticeable by touch in the final print.

[image]

### Project file & How to reproduce

[Dragon_Fly_Phaetus_Spacer.zip](https://github.com/supermerill/SuperSlicer/files/11242684/Dragon_Fly_Phaetus_Spacer.zip)

### Version

2.5.59.2

### Operating system

Windows 11 22H2 - Build 22621.1555

### Printer model

Ender 3 - LGX Lite - Stealthburner - KlackEnder - Phaetus Dragonfly BMS - V6 0.6mm Nozzle

EmreDogann, 2023-04-18:
The XY compensation is from elephant's foot unfortunately so I have to use it. I will try these settings though thanks! I suspect these Arachne settings will lead to some overextrusion correct? I guess in that case I can put down the flow ratio a bit.

EmreDogann, 2023-04-24:
It does help a little bit, but unless I use the aggressive settings suggested by Dx, the gaps are still there. I get some overextrusion with those settings though so I have lowered the flow a bit.

inspectionsbybob, 2023-04-24:
What issue are you trying to solve?  The Model looks good, and I suspect it will be very solid as is with no artifacts showing on the finished surfaces. Small voids like that shouldn't affect it much unless I am missing something.

inspectionsbybob, 2023-04-24:
For top surface, I increase my top layers and turn on Ironing.

3 top layers is about my minimum if I want a pretty top, 5 is better.  Ironing and Monotonic together make a nice top layer.

I am not sure Arachne has been fully polished yet.  It was supposed to be the fix for gap-fill, but I have seen lots of comments like your about "almost but not quite"....  

What I read a while ago was that it was supposed to dynamically change the line width to fill in gaps like yours,  sort of a horizontal dynamic layer height.

EmreDogann, 2023-04-24:
Sorry, I should have also mentioned that the gaps appear between perimeter walls where there is no infill (like on the rim of the boat of the benchy). You're right, for top infill ironing could be a quick fix. I agree though I think it might need a little more time in the oven before it can fully replace classic slicing + Gap fill. For functional prints though it can really speed things up allowing you to use larger nozzle sizes with almost comparable quality which is what I've been mainly using Arachne for.

closed 2023-04-24 by EmreDogann