supermerill/SuperSlicer #3798: EXCLUDE OBJECT causes mcu 'mcu' shutdown: timer too close in Klipper

NiiXXiiN, 2023-04-16:
### What happened?

Version of Superslicer 2.5.59.2 exports some kind of polygon files if i read correctly the gcode.
That causes the pi to get overwhelmed and shut  down.
When slicing the exact same file in Superslicer 2.5.59.0 the file gets printed correctly without any shutdowns.
The project i get this problem with has 105 tiles of approximate dimensions 20x20x3mm on a 300x300mm buildplate.

### Project file & How to reproduce

[Example File For Bug Report.zip](https://github.com/supermerill/SuperSlicer/files/11244820/Example.File.For.Bug.Report.zip)

### Version

2.5.59.2

### Operating system

windows 11

### Printer model

v-core 3

NameOfTheDragon, 2023-04-27:
Is this really a slicer issue or a Klipper/moonraker/mainsail issue? Have you tried turning off `Print Settings` -> `Output Settings` -> `Label Objects`?

NiiXXiiN, 2023-04-27:
> Is this really a slicer issue or a Klipper/moonraker/mainsail issue? Have you tried turning off `Print Settings` -> `Output Settings` -> `Label Objects`?

As described before it's a slicer issue. The previous version of superslicer works fine with the same settings.

NiiXXiiN, 2023-11-27:
Up, for review by @supermerill

closed 2024-02-11 by supermerill

Tehvan, 2024-06-07:
This bug may still be present. It seems to happen when I have a lot of objects I am trying to print. Ah, I took a look at your commit. Yes, I am trying to print over 20 objects.

NiiXXiiN, 2024-06-07:
@Tehvan it got fixed for me after supermill's fix on the newer versions.