supermerill/SuperSlicer #3799: SS insists on slicing on a move? (ie, how do I turn off auto-slice?)

inspectionsbybob, 2023-04-16:
### Version
3.5.59

### Operating system type + version
Win11

### Behavior
MANY times, after slicing I want to just move an object because it is too close to another object or an edge 
(or a bad spot on the plate)
As soon as I move the object SS invokes a "Slice Now" which locks me out of moving another piece until it is done. (or cancel and switch back to the bed from preview)

HOW DO I TURN THIS OFF so it only slices when I want it to?

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

WickedInvi, 2023-04-17:
Stop the slicing first before making changes or press ctrl + P and pick Don't Switch from the drop-down. This will still slice but will stop switching to the slice/gcode view
[image]

neophyl, 2023-04-17:
Bob thats the background processing tickbox option.  With that enabled it will slice on each change/move etc. With background processing OFF then you will need to use the slice now button or the keyboard shortcut.

inspectionsbybob, 2023-04-17:
I definitely DON'T have that box ticked as it is a royal PAIN when you have lots of edits.

So there is a BUG somewhere that thinks its enabled.

neophyl, 2023-04-18:
Bob can you use Help > Show configuration folder and find your SuperSlicer.ini file.  The zip it up and attach it.  Maybe someone can find something in it.

inspectionsbybob, 2023-04-18:
Here is the INI file along with confirmation that was not selected

[image]
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/11262601/SuperSlicer.zip)

inspectionsbybob, 2023-04-19:
> Bob I couldn't see any problems with the contents of the ini file. I was
> kinda hoping there would be a duplicate of the setting but there isn't and
> its set to 0 (off).
>
> Grasping at straws but is there any other copies of the software and ini
> files located elsewhere on your computer ? That setting is a really basic
> one thats been in there forever which is why I'm surprised its playing up.

Nope, this is a clean computer (due to a crash) so there is only one
install on it.

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
***@***.***
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

supermerill, 2023-11-22:
hello
It's been a time, but i can't reproduce the error.
The only way to trigger a slice when you have not the background processing, should be moving to the preview/gcode view.
Can you zip and send me your config directory (or at least the root files) ? zip the directory you can find by `Help->Show configuration folder` ? To help me replicate.
Also, if you can make me a list of steps to reproduce (what to click after opening the slicer)
Thanks.

inspectionsbybob, 2023-11-22:
Sorry so long ago I am not using that configuration anymore.  I think it is in the base code as I am seeing it in Prusa slicer. I am using that for the time being because of some desired features not yet ported over…

supermerill, 2023-11-22:
np
Feel free to re-open if this issue araise again.

closed 2023-11-22 by supermerill