supermerill/SuperSlicer #3803: Extrude Loop: wipe_inside_start

NerdHarder615, 2023-04-21:
### Version
2.5.59.2

### Operating system type + version
Linux

### Behavior
Slicing 2 files and getting an error "Error while writing gcode: two points are at the same position. Please send the .3mf to the dev team."

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
File is 36MB so I can't upload it here, this is a link to the file:
Zip:
https://drive.proton.me/urls/87DBASPTZW#9UQJsxQwqakM
tar.gz:
https://drive.proton.me/urls/RNY685FK0G#kUGOzISGdVv9

STL Files:
https://drive.proton.me/urls/KFDFYS4EC4#0aOMOSgfULT4

legoman27, 2024-05-24:
Version - 2.5.59.8
OS - Win 10 home 10.0.19045

had the same issue slicing a file

STL has some errors that may or may not be related
auto repair says it did the following;
4 edge fixed
1 facet removed
22 open edges remaining

the file slices successfully when 'Print Settings > perimeters & shell > overhangs > threshold for' is set to FALSE

3mf project file
https://drive.google.com/file/d/1ILJbISQ8Ko8ZIGQi95BD_MKwFxLs6Ezd/view?usp=sharing

supermerill, 2024-05-27:
Thanks for the report.
It's an arachne bug, fixed in latest release

closed 2024-06-05 by supermerill