supermerill/SuperSlicer #3810: When slicing, many thousands of notch.svg files are created in my application folder.

benfinklea, 2023-04-24:
### What happened?

When I click the "Slice now" button, just after the slice is finished, many thousands of 1kb notch.svg files are created in the Applications folder. They are numbered but all end in notch.svg. For example:

0_1_210_Nnotch.svg
5_1_215_notch.svg
5_1_215_Nnotch.svg
4_1_214_notch.svg
4_1_214_Nnotch.svg
3_1_213_notch.svg
3_1_213_Nnotch.svg
2_1_212_notch.svg
2_1_212_Nnotch.svg
1_1_211_notch.svg
1_1_211_Nnotch.svg
0_1_210_notch.svg

### Project file & How to reproduce

1. click on "Slice now".
[project for bug report.3mf.zip](https://github.com/supermerill/SuperSlicer/files/11305987/project.for.bug.report.3mf.zip)

### Version

2.5.59.2

### Operating system

macOS 13.3.1

### Printer model

Ratrig V-core 3 400^3

HateWhatOwnsYou, 2023-05-11:
same issue here on windows, my superslicer folder on one drive is 625mb lol

bitingsock, 2023-05-11:
In the mean time, if you don't need it, you can turn off `Perimeters & Shell>Advanced>Seam Notch` to prevent this.

VanessaE, 2023-07-22:
I can confirm this here also.  Worse is that the files are being dumped into my home dir instead of some folder in /tmp

VanessaE, 2023-07-22:
In the meantime, since those SVGs are just being dumped into whatever is the current directory when SuperSlicer starts, Linux users can work around this problem by running the program from a script like this:

```bash
#!/bin/bash

notch_svgs="/tmp/Superslicer-notch"

rm -rf $notch_svgs
mkdir -p $notch_svgs
cd $notch_svgs

/data/Archives/Software/Linux/SuperSlicer-ubuntu_18.04-2.5.59.2.AppImage
```

This will confine those SVG's to a more suitable place.  Or at least it seems to work.  Obviously that last line is specific to my system.