supermerill/SuperSlicer #3811: Treat object as Support

inspectionsbybob, 2023-04-24:
**Is your feature request related to a problem? Please describe.**
I routinely try to print very (very)  tall spindly things like axles.  They need "special" support.

In Cura I can build "Buttress" style supports and then tell the slicer to treat them as supports, which means enforcing the X/Y/Z distances.

**Describe the solution you'd like**
I would like to be able to click on any object and tell the slicer to treat it as a support.

**Describe how it would work**
See above

**Describe alternatives you've considered**
Using a different slicer (Cura) allowed me to print the object

**Additional context**
A nice cura feature.

The attached Zip file has a typical axel in the middle with three support posts resting against it.  In cura I can select the diagonal posts and treat them as supports, as in I can specify a .3mm spacing.  This keeps it still so it won't do weird things as it gets taller.
[axel.zip](https://github.com/supermerill/SuperSlicer/files/11312052/axel.zip)

imkebe, 2023-04-26:
Putting aside FR. I managed to print ~30cm height rods just with proper brim and z-hop for 10-20% of the last layers.

inspectionsbybob, 2023-04-26:
I actually used simplify3d to print them.  No supports of any kind, even under the keys.  S3D just printed them.  Just a brim/raft and print.  They didn't even wobble a bit.  S3D dynamically changed the mm/sec as it got taller...
There is a reason I keep multiple slicers...