supermerill/SuperSlicer #3816: Helical Seams - Discussion and Feedback - Commercial FDM manufacturing of high pressure fittings

thwerks, 2023-04-27:
I'm now nearly 3 years into my research and development of the commercial manufacturing of FDM printed pipe couplings and threaded adapters and many other related parts. I have over 250 different models and printed thousands parts and have dove way more into making water tight printed parts than i ever thought i needed to. Some of the parts can hold up to 400psi.

Overall there seems to be a common theme among settings needed to pull this off consistently.  Retraction is generally to be completely avoided; it tends to suck up air in-between the nozzle and heat break and create bubbles upon deretractions that cause leaks.  

Far more concerning than retractions though is how to address the seam, the main source of failure in making high pressure air tight parts. A straight seam is totally out of the question, repeated overlapping stop/starts not only destroys the threads it makes for a leaky part. Random seams are not any better because the nozzle whips around wildly which disturbs the molten plastic and causes leaks.  

What the Solution? A Helical seam.  This type of seam is somehow created accidentally when the seam is set to nearest, you enable the "Wipe Inside" printer settings of "Depth: %100" and "at end" . This however only works in SuperSlicer version 2.4.58.5, all newer versions since then dont product a helical or concessively shifted seam. 

[image]

[image]

In all later versions of SuperSlicer is get this:

[image]

The best i can do in all newer versions of SuperSlicer is to enable 1mm or retraction to trick the slicer into performing a wipe move to get the seam line to shift over by a certain amount with each layer.

[image]

So this brings me to my point, it seems reasonably obvious that having the seam line shift over by a certain (small) amount concessively with each layer is a superior way of making threaded parts and air tight parts. Random and scattered both have many undesirable side effects: increases stringing and print times due to overly long travel moves, more likely to knock part off the bed, more likely to disturb the previous layer and ooze, still may have clumps of seams in a certain area. 

I'm suggesting the there be an option to have a seam line that can shift by a certain specifiable amount and or have it work in conjunction with the "nearest" seam option; whereby if circular then its helical but if has a hard edge then it can be straight.

[image]

im already getting this behavior but it would be nice not to have to use a hack like wiping or sudo retraction to get it. It would be nice to see a helical seam option. There are a lat of use cases where a helical seam is better than the other options.

AssiveMass, 2023-05-02:
Good stuff. I'd also like to see this "Line on gap" feature in Superslicer. I doubt it will be implemented to vanilla Prusaslicer.
https://github.com/prusa3d/PrusaSlicer/issues/9915

Santoshsomi, 2023-05-15:
@thwerks you are saying that creating seam at same place for inner and outerwall can make part weak ? 
[image]
Something like this doesnt work? Wipe does happen at outerwall for 1/2 and next 1/2 length of wipe happens inside. But seam is at single point.  Just want to understand your thoughts.

thwerks, 2023-05-15:
I'm saying that in 3 years of testing high pressure pipe fittings, vertical seam lines seem to be a lot less reliable in terms of being air/water tight. As far as mechanical strength, i doubt there's any difference. Shifting the seam/wipe/retraction sequence by a certain amount each time seems to more reliably produce airtight parts. 

And think about it in terms of threads meshing together in the mating of two pipe fittings. Having the seam sequence in the same area will make the meshing of threads lopsided; that alone makes it leak prone. Adding PTFE tape can't easily compensate for that. Far better to be a random seam in regard to that aspect; though a consecutive shift is better than random in terms of the meshing/seating of threads. 

Consider the  even the subtle difference between that of NPT vs FNPT threads. Small differences can make a huge difference in sealing performance.

Santoshsomi, 2023-05-15:
ok thats sounds fascinating, do you have any picture of printed part ( same as above)? I'm just worried about the blobs if its going shift at a certain amount. That would require perfect retraction and other settings to hide it.