supermerill/SuperSlicer #3817: Vase Mode Crashes Application

NortonTech-Official, 2023-04-27:
### What happened?

When switching to a profile that has vase mode selected or when checking the box to turn on vase mode the application will crash immediately.

### Project file & How to reproduce

[Vase Mode crash.3mf.zip](https://github.com/supermerill/SuperSlicer/files/11348148/Vase.Mode.crash.3mf.zip)

### Version

2.5.59.x

### Operating system

MacOS 12.6.6 (21G633)

### Printer model

Sovol SV01 running Klipper

NortonTech-Official, 2023-04-27:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [4399]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            org.SuperSlicer/
Version:               SuperSlicer SuperSlicer_2.5.59.2_2023-04-25 (SuperSlicer_2.5.59.2_2023-04-25)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-04-27 15:39:48.4341 -0600
OS Version:            macOS 12.6.6 (21G633)
Report Version:        12
Bridge OS Version:     3.0 (14Y910)
Anonymous UUID:        22B6E5C8-ACA2-189C-A06A-7F1116EDDC6F

Sleep/Wake UUID:       CDB73CF8-9815-45B6-A4E1-DC67BA70F8E8

Time Awake Since Boot: 8100 seconds
Time Since Wake:       6126 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000000000000c
Exception Codes:       0x0000000000000001, 0x000000000000000c
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [4399]

VM Region Info: 0xc is not in any region.  Bytes before following region: 140737486807028
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      VM_ALLOCATE              7fffffe86000-7fffffe87000 [    4K] r-x/r-x SM=ALI  

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   SuperSlicer                   	       0x1056d85f9 Slic3r::DynamicPrintConfig::normalize_fdm() + 1753
1   SuperSlicer                   	       0x105672aca Slic3r::Print::apply(Slic3r::Model const&, Slic3r::DynamicPrintConfig) + 442
2   SuperSlicer                   	       0x105b58303 Slic3r::BackgroundSlicingProcess::apply(Slic3r::Model const&, Slic3r::DynamicPrintConfig const&, Slic3r::DynamicPrintConfig const*) + 163
3   SuperSlicer                   	       0x105da4837 Slic3r::GUI::Plater::priv::update_background_process(bool, bool) + 215
4   SuperSlicer                   	       0x105dd6735 wxEventFunctorFunctor<wxEventTypeTag<wxTimerEvent>, Slic3r::GUI::Plater::priv::priv(Slic3r::GUI::Plater*, Slic3r::GUI::MainFrame*)::$_19>::operator()(wxEvtHandler*, wxEvent&) + 37
5   SuperSlicer                   	       0x107f56fae wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
6   SuperSlicer                   	       0x107f56d1b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
7   SuperSlicer                   	       0x107f56bd9 wxEvtHandler::ProcessEvent(wxEvent&) + 41
8   SuperSlicer                   	       0x107f5714c wxEvtHandler::SafelyProcessEvent(wxEvent&) + 12
9   SuperSlicer                   	       0x107f37b1a wxTimerImpl::SendEvent() + 90
10  CoreFoundation                	    0x7ff809266f49 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 20
11  CoreFoundation                	    0x7ff809266a38 __CFRunLoopDoTimer + 923
12  CoreFoundation                	    0x7ff8092665a8 __CFRunLoopDoTimers + 307
13  CoreFoundation                	    0x7ff80924ccb6 __CFRunLoopRun + 2010
14  CoreFoundation                	    0x7ff80924be1c CFRunLoopRunSpecific + 562
15  HIToolbox                     	    0x7ff811efb5e6 RunCurrentEventLoopInMode + 292
16  HIToolbox                     	    0x7ff811efb34a ReceiveNextEventCommon + 594
17  HIToolbox                     	    0x7ff811efb0e5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
18  AppKit                        	    0x7ff80bc85f6d _DPSNextEvent + 927
19  AppKit                        	    0x7ff80bc8462a -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394
20  AppKit                        	    0x7ff80bc76cd9 -[NSApplication run] + 586
21  SuperSlicer                   	       0x107e7747e wxGUIEventLoop::OSXDoRun() + 174
22  SuperSlicer                   	       0x107f60dfd wxCFEventLoop::DoRun() + 29
23  SuperSlicer                   	       0x107ed65e5 wxEventLoopBase::Run() + 85
24  SuperSlicer                   	       0x107eaee14 wxAppConsoleBase::MainLoop() + 116
25  SuperSlicer                   	       0x107e3c66a wxApp::OnRun() + 26
26  SuperSlicer                   	       0x107f09563 wxEntry(int&, wchar_t**) + 131
27  SuperSlicer                   	       0x105d2d0df Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 239
[... 274 lines cut ...]
      },
      "deploymentId" : 250000230
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro13,2, BootROM 509.0.0.0.0, 2 processors, Dual-Core Intel Core i5, 2.9 GHz, 8 GB, SMC 2.37f25
Graphics: Intel Iris Graphics 550, Intel Iris Graphics 550, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445364533303445422D45474347202020
Memory Module: BANK 1/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445364533303445422D45474347202020
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x157), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1710.4)
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Network Service: Tailscale Tunnel, VPN (io.tailscale.ipn.macos), utun3
USB Device: USB30Bus
USB Device: SteelSeries Aerox 3 Wireless
USB Device: Apple T1 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.5

DarKingF0x, 2023-04-29:
I'm also having this problem. I thought it was my pc at first so I uninstalled and reinstalled the same version and vase mode continuously crashes the app.

timschneider, 2023-05-02:
duplicate of #3749 #3731 #3722 #3715 fixed with #3736

closed 2023-11-22 by supermerill