supermerill/SuperSlicer #3821: Acceleration Control Not Correctly "Decel w/ Target" [RRF]

CCS86, 2023-04-30:
### What happened?

When acceleration control is used and "decelerate with target acceleration" enabled, the gcode is incorrect:

[image]

Despite splitting the travel moves to allow the target decel, no M204 command is given between, so the travel move decelerates at the full 5000 mm/s^2, instead of the desired 1200.

### Project file & How to reproduce

[decel-to-target.3mf.txt](https://github.com/supermerill/SuperSlicer/files/11360447/decel-to-target.3mf.txt)

### Version

2.5.59.2

### Operating system

Windows 10 x64

### Printer model

UMO+

CCS86, 2023-07-26:
[image]

CCS86, 2023-12-24:
@supermerill

supermerill, 2023-12-29:
fixed.
It was only checking the main accel to check if it write it.
but in your first picture, it's 3650 as it began by the internal perimeter.

closed 2024-02-11 by supermerill