supermerill/SuperSlicer #3829: Support for Auxiliary Fan

lettore, 2023-05-02:
As you know Bambulab introduced the Aux Fan for help cooling the material when printing at high speed.
The fork of their slicer, Orcaslicer, have this useful feature, it would be possible to add this to Superslicer?
[image]
[image]
Orcaslicer/Bambu Studio send a `M106 P2 xxx` Gcode where P2 is the Aux Fan.
At the moment I added this command at the Start and End Gcode section of the filament, but in this way the fan will start at the print beginning and is not really good for the first layer.
It would be also amazing if in the future it can be activated only for the overhangs for example.

Max-Luck, 2023-05-04:
Hi!

While that feature can exist, since you are using klipper you can add this feature by using custom variables feature under Printer Settings -> Notes -> Custom Variables. 

From there you can pass the variable to a klipper g-code macro to get the desired effect. 

Ellis talks about it here https://ellis3dp.com/Print-Tuning-Guide/articles/passing_slicer_variables.html

I would just nest this behavior in your M190 like in this macro https://github.com/VoronDesign/VoronUsers/blob/master/printer_mods/Ellis/Bed_Fans/Klipper_Macros/bedfans.cfg

If this sounds like a lot of work you can also try adding this to your "After Layer Change G-code" and using the [layer_z] to define which layer you want the fan to kick on at
[image]

lettore, 2023-05-04:
Hi,
yes it's absolutely possible to make it work as you say, with macro there's an infinite ways to achieve that.
Anyway I think it would be better if the slicer can do that directly, first because in this way you have to change the printer settings while usually the aux fan is used only in PLA, so it should be a setting visible under the cooling section of the filament, without messing up with gcode settings section. Second because with the time it's possible to expand the logic of when using the aux fan, maybe only on overhangs or bridges?

legend069, 2023-06-24:
I'll support this feature. But I'll need to look more into it, will need to see how klipper/marlin/bamboo support fans

Currently you can do this in the custom gcode area.