supermerill/SuperSlicer #3830: Jagged slicing

Packman65, 2023-05-03:
### What happened?

I started trying SuperSlicer 2.5.59.2 a week ago. Initially, it would work predictably and I managed to slice models and get matching prints. What I previewed in SuperSlicer is what was printed out. Fine.
Tonight, I sliced a new model, and what I saw in the preview shocked me. Well, initially I was amused for a short while, thinking I had perhaps turned on some sort of 'fuzzy skin' feature. I did not find any such feature in the print settings. Please view attached images. I closed everything and rebooted the laptop, and jagged sliced result was still present. Just to check that this behavior was not due to the model I sliced, I loaded a model (BRT3.STL, Bridge Roof Test) I had sliced successfully a week earlier. Now, this came out jagged too. I've saved it as a 3mf and it's included here. No, I have not tried printing these jagged slice jobs.
Is SuperSlicer self-deteriorating with added use? Why I wonder about this is previously, I tried version 2.4.x.x to slice BRT3.STL. Initially I got the bridging to work the way I wanted by adjusting the settings, but later on as I sliced the same model with the same settings, I noticed the prints getting worse again (bridges again failing to reach perimeters and dropping in mid-air). Actually, that is why I ended up trying 2.5.59.2.
I see hope in SuperSlicer becoming my go-to slicer, as I find it combining what I like in Cura and PrusaSlicer, with effective bridging settings which I find missing or not working well in Cura. Thus, I'm hoping SuperSlicer will be stable in the long run. Thanks. 
[image]
[image]
[image]
[image]
[image]

### Project file & How to reproduce

[BRT3.zip](https://github.com/supermerill/SuperSlicer/files/11384324/BRT3.zip)

### Version

2.5.59.2

### Operating system

Windows 10

### Printer model

SeeMeCNC Rostock Max v3

neophyl, 2023-05-03:
Try turning fuzzy skin OFF.  When I open your project in 2.5.59.2 you have fuzzy skin set for all perimeters  150% thickness and 200% distance.
[image]

Packman65, 2023-05-05:
neophyl, you're right! My bad.
I didn't even know I turned on fuzzy skin. I hadn't even noted where that was till now. I'm pretty new to SS, having been mostly a  Cura user till recently. To slice the first model mentioned, I had only changed the number of bottom layers, and increased infill percentage. I think I know how I accidentally turned on fuzzy skin. My mouse pointer must have been over that setting when I rolled my mouse scroll-wheel, since the 'Perimeters & Shell' view is taller than can fit on my screen. I've tested, and this will cause the pull-down options to rotate through and it must have been left at "All perimeters" without my realizing it. Lesson learnt, I'll make sure to keep my pointer to the right of the screen when scrolling.

Along this topic of settings, I've found that the "Seam position" setting in 2.5.59.2 does not allow the first two options, "Corners" and "Nearest" to be selected. Selecting any of these two will result in it changing to "Custom". I wonder if this is a bug, and if so, whether it is already being worked on.

Oh, BTW, I have SuperSlicer 2.4.x..x and 2.5.59.2 on separate PCs.

Thanks for looking into this case, neophyl.

vkhodygo, 2023-05-06:
@Packman65 do you mind closing this issue as resolved?

closed 2023-05-07 by Packman65