supermerill/SuperSlicer #3831: Gap fill - not constant

stankosik, 2023-05-03:
### What happened?

Hi everyone.
I have a problem with gap fill, and more specifically with filling along the entire length. Gap fill is not constant. It changes - it is larger and smaller along the entire length of the filling.
I tried different width and height settings... and still the same.
I even checked on Ellis-SuperSlicer-Profiles - Ultra High Flow Hotend + CHT (44mm3) and it's exactly the same.
The condition is: 0.8mm nozzle and 0.5mm layer height
Can anyone help?

[image]

### Project file & How to reproduce

[Wall2 - test.3mf.zip](https://github.com/supermerill/SuperSlicer/files/11390468/Wall2.-.test.3mf.zip)

### Version

2.4.58.5

### Operating system

macOS Ventura 13.2.1

### Printer model

Modix Big 60 v3 + Takoto Monster 0.8

Max-Luck, 2023-05-03:
Hi!

The problem isn't with the slicer, but how the model was exported. The model is exported with a low facet count. Meaning it is squared/triangled off in areas to reduce file size. See the photo below, you might have to adjust your monitor or import it into a program where you can adjust the contrast ratio. So the actual file varies its width along the curve. The slicer is trying to compensate for that. 

You can try to negate this effect by re-exporting the file at a higher polygon count (if you are using Fusion 360, in the export to mesh change the refinement level to high), or using slicer settings to negate this (specifically I would look at adjusting your internal perimeter width to close this gap)

I also imported the file into Fusion 360 so you can see how the file is triangulating. 
[image]
[image]

stankosik, 2023-05-04:
I did as You wrote and it is perfect.
I changed in Refinement Options:
Surface Deviation to minimum
Normal Deviation to 1 
Thank You for your reply and for your advice.

Phil1988, 2023-05-12:
Is this solved? if yes, do supermerill a favor and close it :)

closed 2023-05-12 by stankosik