supermerill/SuperSlicer #3839: Merge PrusaSlicer support for different SLA formats

aelnagger, 2023-05-12:
**Is your feature request related to a problem? Please describe.**
Prusaslicer recently merged functionality to support writing sliced models for anycubic mono and mono SE in https://github.com/prusa3d/PrusaSlicer/pull/9929. Sometime last year prusaslicer renamed `SLAArchive` to `SLAArchiveWriter` and introduced several other changes to support a wider variety of sla output formats. It would be helpful to try to pick up those changes to better support SLA formats like the CTB format requested in #1488.

**Describe the solution you'd like**
Bring the SLA formatting modules up to speed with the changes implemented in prusaslicer.

**Describe how it would work**
Not having a full understanding of how functionality from prusaslicer usually gets merged into SuperSlicer, the most straightforward approach to me seems to be to look at the relevant pull requests and try to cherry-pick the resulting merge commits.

**Describe alternatives you've considered**
The various files make up the SLA formatting logic could also be copy/pasta'd. That reduces the need to painstakingly resolve merges, but also loses the git history of the upstream changes.

**Additional context**
https://github.com/prusa3d/PrusaSlicer/pull/9929

aelnagger, 2023-05-12:
I'm comfortable making an attempt at this.

legend069, 2024-01-24:
i would recommend waiting for merill to finish the 2.7 merge before starting on this.
there already is a fair few bits of code for SLA slicing in 2.4.58.5 so you should be able to start with cherrypicking a few commits at a time and fixing any issues that they might cause