supermerill/SuperSlicer #3840: Unlinked perimeters

Scrambler-77, 2023-05-12:
### What happened?

Archne does not link external perimeters as shown in pictures.
Tried all sort of setting without success.
[image]
[image]

### Project file & How to reproduce

[TEST.zip](https://github.com/supermerill/SuperSlicer/files/11465281/TEST.zip)

### Version

2.5.59.0 and 2.5.59.2

### Operating system

Windows11

### Printer model

RatRig VCore3.1

neophyl, 2023-05-12:
You  must not have changed many settings.  
[image]

Thats with the Wall transition set to 20 instead of 10 degrees.
[image]

(With 2.5.59.2)

Scrambler-77, 2023-05-16:
Many thanks! I still cannot understand well the exact meaning of those arachne options.