supermerill/SuperSlicer #3842: Search lists the same item multiple times, and mousing over that item changes what it is

asolidtime, 2023-05-13:
### What happened?

This is pretty much impossible to describe in text, so I just recorded a video: [https://youtu.be/CX5mbs3FY6o](https://youtu.be/CX5mbs3FY6o)

### Project file & How to reproduce

1. Open the search window with Ctrl+F
2. Search for something

### Operating system

Arch linux (rolling)

### Printer model

ender 3, using the default profile

trysten, 2023-05-26:
I'm also getting this on Debian testing - Sway/wlroots. To be fair, I get a few strange bugs like this that I blame on wayland.

VanessaE, 2023-08-21:
I can reproduce this also, using the 2.5.59.0 appimage under Debian Bullseye.

supermerill, 2023-11-22:
will test on my linux.