supermerill/SuperSlicer #3848: Crash when slicing, related to solid infill using Concentric (filled)

Overruler, 2023-05-16:
### What happened?

SuperSlicer crashes to desktop when slicing this project file. If the option Infill->Solid is changed to Rectilinear, slicing completes without crashing. Problem details as recorded by Windows:

Source
SuperSlicer

Summary
Stopped working

Date
‎2023-‎05-‎16 13:33

Status
Report sent

Description
Faulting Application Path:	D:\SuperSlicer_nightly\superslicer.exe

Problem signature
Problem Event Name:	APPCRASH
Application Name:	superslicer.exe
Application Version:	2.5.59.2
Application Timestamp:	64487510
Fault Module Name:	Slic3r.dll
Fault Module Version:	0.0.0.0
Fault Module Timestamp:	64487504
Exception Code:	c0000005
Exception Offset:	00000000000a8d74
OS Version:	10.0.19045.2.0.0.256.48
Locale ID:	3072
Additional Information 1:	b4cc
Additional Information 2:	b4cc2d92f094399daa90dac9d7fc9c60
Additional Information 3:	c8ac
Additional Information 4:	c8acfdf717ae3a8103b66626d53db0a7

Extra information about the problem
Bucket ID:	fe47913feddd0d2dad1eb9c8e132e058 (2098318748824035416)

### Project file & How to reproduce

[DeskLampBase_bug.zip](https://github.com/supermerill/SuperSlicer/files/11487504/DeskLampBase_bug.zip)

To reproduce, click 'Slice now'.

### Version

2.5.59.2

### Operating system

Windows 10 Version 22H2 (OS Build 19045.2965)

### Printer model

Original Prusa i3 MK3S & MK3S+

vampyrex13, 2023-07-17:
Confirming same issue as of build 626 up to 631. Even just having Conentric (filled) selected without slicing will cause SuperSlicer to crash.

vampyrex13, 2023-11-23:
I just tested build #637 with all my files that crashed before, and concentric infill models now slice without crashing.

closed 2024-02-11 by supermerill