supermerill/SuperSlicer #3851: Add support for detecting "Top side" overhangs

lettore, 2023-05-17:
I'm printing many double wall objects without infill and I'm often incurring in the problem where top surfaces angled more than 45° aren't printed well.
[image]
Detail of the top view
[image]
This is the bottom view
[image]
The geometry is the same but the difference is that Superslicer correctly detect the bottom as a Overhang and slow down the print speed accordingly but doesn't the same on the upper face.
[image]
Looking at the sliced section it's obvious that the inner wall is an ovehang, and with that circular shape the extrusion tend to be pulled inside the print because it have not enough time to stick to the previous layer.
So I think it will be useful to detect not only "negative" overhangs ( with negative I mean the classic ones) but also "positive" overhangs.

neophyl, 2023-05-18:
What settings do you have for Ensure Vertical Shell thickness and no solid infill over ?  Without an attached project as requested its very hard to tell if it could be handled with a settings change or if there is an actual need for a software change.

martian099, 2024-01-24:
Same problem here. Waiting for a fix

supermerill, 2024-01-24:
I'm not sure what you want.
Like neophil said, using the 'Ensure Vertical Shell thickness' and a bigger 'no solid infill over', there is infill put where you need it.

you don't want the infill, but still something to fill the gap?

Fabface, 2024-01-24:
I assume the "problem" will go away if you import your model as a shell, because the model itself does not actually have overhangs where you want them to be detected - these overhangs only exist because you print without infill. If you import a hollow model, it will be detected correctly.

lettore, 2024-01-24:
> I assume the "problem" will go away if you import your model as a shell, because the model itself does not actually have overhangs where you want them to be detected - these overhangs only exist because you print without infill. If you import a hollow model, it will be detected correctly.

Yes but actually if I print without infill overhangs exist also "inside" the closed shell. So, as I said ,there's should be an option do enable overhang detection also on the inside of the print, from my point of view top surfaces with more than X degrees angle. 
Making an hollow version of the object obviously not work as it needs to be closed

supermerill, 2024-01-24:
ok, I understand.
when there is 0% infill, detecting overhangs over infill.
mmm.
Not that straitforward. Currently, periemters are computed in parallel. But here, you need to know where the previous perimeters are.
So I need to change the workflow with this new setting enabled, so I can get the infill part of the previous layer, to use it as a 'not supported' area.

martian099, 2024-01-25:
Thank you for looking into it! Here are more similar posts that explains this issue better than I could: https://forum.bambulab.com/t/detect-internal-overhangs/45239, https://forum.prusa3d.com/forum/prusaslicer/internal-overhang-and-bridging-speeds/