supermerill/SuperSlicer #3854: Feature request: Flow rate compensation for non-linear extruder behaviour

tehdusk, 2023-05-22:
**Is your feature request related to a problem? Please describe.**
Usually less filament is extruded at higher extrusion speeds even though the same amount is requested.
This does not only raise issues on very high speed prints but also systematically introduces flow inconsistencies at moderate flow rates, leading to over-extrusion at very slow print speeds and under extrusion at higher speeds.
A much more detailed description can be found at https://www.cnckitchen.com/blog/extrusion-system-benchmark-tool-for-fast-prints

**Describe the solution you'd like**
A lookup table for filament/extruder combo can easily be created by the user at home using a fine scale. Here is an example.
[image]
When it's expected that actual flow will be lower than originally requested, increase demanded flow so the target is met.
flow_commanded = flow_requested * (1/flow_expected)

**Describe how it would work**
If for example a flow drop of 3% is expected, add the required amount to reach the target.
In this example, using the formula above with flow_expected = 1-0.03 = 0.97
flow_commanded = flow_requested * (1/(0.97)) = flow_requested * 1.031

**Describe alternatives you've considered**
For the most part during the usable extruder range, flow drops somewhat linear. A simpler implementation could use a base flow rate at a lower speed (as is now) and have a linear drop-factor.
For example in the above graph I'd set 5 mm³/s as 100% and set the expected drop to 1% per 10 mm³/s flow rate increase. In addition to that, I'd set 35 mm³/s as the maximum allowable extrusion rate (already implemented).

vkhodygo, 2023-06-05:
That's a nice one, but you need more measurements per point to make sure these values are actual. How much filament did you use for this figure?

vkhodygo, 2023-06-08:
Cool! Don't you think this should be a firmware feature in this case?

Obliman, 2023-06-12:
It sounds like this could help address some other issues people have been having where small areas (lower speed due to many accelerations and decelerations) tend to be over-extruded when the extrusion multiplier is calibrated for larger areas (higher average speed).

TheBoojah, 2024-03-06:
@tehdusk There is a pull request with this feature over at prusaslicer: https://github.com/prusa3d/PrusaSlicer/pull/12186
Hopefully this could be implemented in superslicer first for testing by people willing to tinker with calibration and settings!

> It sounds like this could help address some other issues people have been having where small areas (lower speed due to many accelerations and decelerations) tend to be over-extruded when the extrusion multiplier is calibrated for larger areas (higher average speed).

Exactly what I have been experiencing, while desperately trying to get perfect top layers with ironing I could never get both large areas and smaller sections to be equally filled. It is very noticeable that when the extruder speeds up in the middle of a larger area, the flow is reduced with quite visible under extrusion as a result.

TheBoojah, 2024-06-27:
Has this feature been in there for a long time?
[image]
Seen in the nightly as of today.
I'm going to test it soon to see if it solves my issues! Thanks!

supermerill, 2024-06-28:
it's very new. It needs to be reviewed.

tehdusk, 2024-07-20:
Great that this has been added. I did a simple test keeping most of the variables constant and under somewhat real world conditions. In short, the theory seems to work as expected in practice but the implementation can be improved.

I printed 3 discs which should be around 14.6 g each. With the print settings I use on a .5 nozzle and .25 layer height varying the speed 50/100/150 mm/s results in extrusion rates of 7/14/21 mm³/s.

[image]

So here's the data:

|     ExMultiplier(v)    | 50 mm/s  | 100 mm/s  | 200 mm/s |
| ------------------- | ---------- | ----------- | ----------- |
| 1.000/1.000/1.000 | 14.587 g  |  14.478 g   |  14.356 g  | 
| 1.000/1.008/1.017 | 14.592 g  |  14.587 g   |  14.597 g  | 

Raising the extrusion speed with a constant multiplier leads to underextrusion, which is no longer present with extrusion multipliers dependent of extrusion speed. When scratching the surface with a fingernail the underextrusion on the top layer at 200 mm/s is also gone (with adjusted extrusion multipliers) and the samples printed at 200 mm/s can not be distinguished from the 50 mm/s samples anymore. So that's exactly the desired outcome.
However, I think print speed is not a good choice for the x-axis because the flow changes significantly depending on the width of the extrusion and of course the layer height. Flow rate (mm³/s) or extruder speed (mm/s) should better correlate with actual extrusion.
In my experience slippage in the extruder also depends on the filament. So filament settings might be a better place, making it more universal.

Sometimes when playing around with it I get strange slicing results... not sure if this is due to the feature or something else. Note the disc with 200 mm/s is missing above the first layer in that case, probably something with inserting the exMultipliers went wrong... that interface doesnt save sometimes, and adjusting the min/max flow & speed also seems buggy.
[image]
And this should be a temp tower from the calibration menu, where the faster extrusions are also missing.
[image]

supermerill, 2025-07-19:
> +1 to that. And this statement (to use flow rate instead of print speed) is aligned with the initial study by Stefan linked in the issue description above. He uses `flow ratio` as well

I can do that, Very easy as it's almost exactly the same thing.