supermerill/SuperSlicer #3856: Label Objects Not Working for Klipper

RickoT, 2023-05-24:
### What happened?

EXCLUDE_OBJECT_DEFINE lines are missing from gcode

### Project file & How to reproduce

Select **Klipper** from **Printer Settings** > **General** > **G-Code Flavor**
Check the **Label objects** checkbox in **Print Settings** -> **Output Options** -> **Output FIle**

Slice and save a G-Code file with at least 2 objects
Edit the G-Code file with notepad or some other application

Note there is no lines containing EXCLUDE_OBJECT_DEFINE 

Alternatively, print the gcode file with a klipper install containing [exclude_objects] in the printer.cfg file
Note that (in MainSail) the **Exclude Object** button is missing from the dashboard
[xyz-CalbirationCube-ExcludeObjectFailSample.gcode.txt](https://github.com/supermerill/SuperSlicer/files/11550180/xyz-CalbirationCube-ExcludeObjectFailSample.gcode.txt)

### Version

2.4.58.5_win64_220924

### Operating system

Windows 11 22H2 OS Build 22621.1702

### Printer model

Modified Ender 3 running Klipper

salocain, 2023-05-25:
you need a scrit for exclude object

[https://github.com/kageurufu/preprocess_cancellation](url)

work like a charm

RickoT, 2023-05-25:
Actually, I noticed that the EXCLUDE_OBJECT_DEFINE line is there if there's only one object, but not when you have more than one

AudreyAP, 2023-05-30:
Enable postprocessing in Moonraker.

mirzaMk90, 2023-09-07:
> Actually, I noticed that the EXCLUDE_OBJECT_DEFINE line is there if there's only one object, but not when you have more than one

I have the same problem since today. Missing EXCLUDE_OBJECT_DEFINE line, but for me it was there for every object I have on platter, not just if I have one. I don't know what to do