supermerill/SuperSlicer #3857: Crash when dragging and dropping a gcode file into SuperSlicer

RickoT, 2023-05-24:
### What happened?

When dragging and dropping a gcode file into the superslicer main window, the application crashes (see screenshot)

After crashing, the gcode viewer still opens with the preview

[image]

### Project file & How to reproduce

Drag and drop any gcode file into the main SuperSlicer window

### Version

2.4.58.5_win64_220924

### Operating system

Windows 11 22H2 OS Build 22621.1702

### Printer model

Modified Ender 3 running Klipper

supermerill, 2023-11-21:
click continue and it will work.
There is still some quirk with the gui library i don't understand. It complains but still works.

supermerill, 2023-11-21:
seems like a wxwidget error. Will be fixed after updating the library, i'll try again after merging with ps2.7