supermerill/SuperSlicer #3858: Klipper does't recognise gcode with cyrilic alphabet in name of stl file

nightmare04, 2023-05-26:
### What happened?

Klipper does't recognise gcode with cyrilic alphabet in name of stl file

### Project file & How to reproduce

[Tissot belt углубл.zip](https://github.com/supermerill/SuperSlicer/files/11573517/Tissot.belt.zip)

### Version

2.5.59.2

### Operating system

windows 10

### Printer model

Ender 3

neophyl, 2023-05-26:
How is that a SuSi bug ?  Surely thats a klipper issue

Pavulon87, 2023-05-26:
There is an "Illegal characters" setting but it only removes the characters in the filename.
For example. File **łąka.stl** is saved as **____ka.gcode** but inside there is:
> ; object:{"name":"Ĺ‚Ä…ka","id":"Ĺ‚Ä…ka.stl id:0 copy

Creo2005, 2023-05-27:
For klipper - use python3