supermerill/SuperSlicer #3863: Provide an AppImage

tinkerbeast, 2023-05-30:
**Is your feature request related to a problem? Please describe.**
I am using Ubuntu22.04 and I would like an AppImage for it. I spent the last hour install various dependencies and try to compile it natively - However it just doesn't work with native packages (OpenVDB and OpenCascade have compatibility issues). I also found the Linux installation guide to be inadequate.

**Describe the solution you'd like**
I would like and AppImage container for Ubuntu22.04.

**Describe how it would work**
Self explanatory.

**Describe alternatives you've considered**
Native compilation.

**Additional context**
N/A

Pavulon87, 2023-05-31:
Download one of the successful builds available in actions
https://github.com/supermerill/SuperSlicer/actions/runs/4512657331

pedep, 2023-06-01:
AppImages are available for download in the `Assets` section under every release on the [Releases page](https://github.com/supermerill/SuperSlicer/releases)

While the files may be named "ubuntu_18.04", it runs flawlessly on both my Debian Bullseye and Ubuntu Jammy machines. I'm using [this image](https://github.com/supermerill/SuperSlicer/releases/download/2.4.58.5/SuperSlicer-ubuntu_18.04-2.4.58.5.AppImage)

tinkerbeast, 2023-06-07:
I tried using the "ubuntu_18.04" but it crashed on certain instances. For example, when I selected the "Spiral Vase" setting. Since I couldn't get it to run natively,  I am not sure if this is a common bug, or specific to the AppImage build.

Pavulon87, 2023-06-07:
@tinkerbeast, These are not stable builds, try older one.
#3731, #3726, #3710, #3686