supermerill/SuperSlicer #3864: External Perimeter pathing error due to some setting in Custom Print Settings

ChaosBlades, 2023-05-30:
### What happened?

Something in my Print Settings that is not in the default profiles is causing this specific model to have external perimeter pathing errors. No other models that I printed have issues.
[image]
[image]

### Project file & How to reproduce

Something in the Print settings is causing the external perimeters on this model to not follow the models geometry. Tried to change over a dozen settings to narrow down what exactly is causing the issue but I am unable to determine the root cause.
[gcode error project file.zip](https://github.com/supermerill/SuperSlicer/files/11607094/gcode.error.zip)

### Version

2.5.59.2

### Operating system

Pop!_OS 22.04 LTS x86_64 

### Printer model

Ender 3 Pro to Voron Switchwire Conversion

melrhombus, 2023-06-03:
No issue with that model but I have the same issue with another one
[image]
[image]

If I switch layer height from .2mm to .4mm it fixes mine but mangles the other model

phcay, 2023-08-10:
I have the same problem with the profiles of my printers.
I don't use default system profiles because my printers are heavily modified (TronXY X5SA-400, Ender3, TwoTree Bluer).
It is very recurrent, since there are curves on the external perimeters.

However, when I slice your model with one of my profiles with which I encounter this anomaly with my models, curiously, the problem does not appear.

So I performed several tests, and I found the parameters that cause this anomaly to appear.

These are the "Overhang" settings.
For my profile and a model where I encounter the problem, but not with your model, it is "Bridge flow" that is the problem. By passing it from 50% to 80%, for example, the anomaly no longer appears.
For your profile and your model, "Bridge speed and fan" is the problem. Passing it from 55% to 40%, the anomaly no longer appears, in fact it seems to only occur between 52% and 58% for this case.
And in any case the pure and simple deactivation of "threshold for", also solves the problem.

Note that for my profile, the box "threshold for", instead of being checked, it was filled in black. If the value of "Bridge flow" is above 50%, it appears checked again.

Also note that for your profile, the values that are defined for "Bridge flow" and "Bridge speed and fan" are those proposed by default by SuperSlicer.

donkeykong3x3, 2023-09-29:
Thanks phcay! I was able to manually adjust the "Bridge flow" and it worked. Mine was defaulted to 80%, but adjusting it to the SuperSlicer default of 75% fixed it.

closed 2024-02-11 by supermerill