supermerill/SuperSlicer #3866: Crashing when slicing with Overhangs/Extrusion direction/reverse on odd set

claudermilk, 2023-06-02:
### What happened?

Turned on Reverse on odd setting under Perimeters & Shell/Overhangs/Extrusion direction. When slicing, the program crashes.

### Project file & How to reproduce

Go to Perimeters & Shell, Overhangs, Extrusion direction. Turn in Reverse on odd. Slice file.
[Crash Settings.zip](https://github.com/supermerill/SuperSlicer/files/11636257/Crash.Settings.zip)

### Version

2.5.59.2

### Operating system

Windows 10

### Printer model

Voron Trident with Stealthburner and Revo

shuffleznl, 2023-07-02:
For me the `Overhangs: Reverse on odd` and `Quality: Extra Parameters on Overhangs` cannot be set, maybe disabled due to crashing issue?
[image]
[image]

supermerill, 2023-11-20:
cannot reproduce with a cube and your settings, can you export the whole project, one that can crash?