supermerill/SuperSlicer #3869: Max volumetric speed math assumes extrusion multiplier = 1

theodiem, 2023-06-06:
**Bug description**
When using "Max volumetric speed" in either Filament Settings or in Print Settings, the resulting print assumes EM to be 1.
If I set EM to 1, and a max volumetric speed to 20, my print speeds are limited as expected:
[image]

If I don't do any other change, except changing EM to 1.5, for example, the max volumetric speed is "blown" proportionally:
[image]

**Expected behavior**
Limits set by max volumetric speed (in either Filament Settings and in Print Settings) should have EM factor computed when setting the limit. In either examples above, it should have been limited to 20.

**To Reproduce**
Steps to reproduce the behavior:
1. Make sure your print speeds are not limiting the volumetric flow
2. Set Extrusion Multiplier to 1
3. Set Max Volumetric Flow to a known value, for example "10".
4. Generate the G-Code and check the volumetric flow in the gcode preview tab
5. Now only change Extrusion Multiplier to another value, for example "1.5"
6. Check the volumetrric code in the gcode preview and see that the limit produced is multiplied by the EM (which shouldn't, since it shouldn't exceed max volumetric flow).

**Screenshots**
See above.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 2.4.58.5
 - Printer model: Voron 2.4 350mm

CCS86, 2023-06-08:
I guess you could argue this both ways. 

To me, extrusion multiplier is an end of the road correction to get the physical reality of the print to match the calculations of the slicer. With your extruder steps and filament diameter set correctly, you "should" be able to leave this at 1. I find that I need it at 0.95 (like Prusa and others). If having the multiplier set to 1 causes under/over extrusion, technically your real volumetric flow rate is wrong. Tuning the multiplier corrects it to match the nominal flow rate calculation.

theodiem, 2023-06-08:
> 

I understand your point.
The problem happened with me because I use different types of filaments (ABS/PETG) which requires different extrusion multiplier and different maximum volumetric flows not exactly related with extruder steps.
My extruder steps is calibrated correctly, but for example, a tungsten carbide nozzle (full TC, not just an insert) have different internal friction than a nickel plated one. Also PETG likes to stick much more with TC nozzles than nickel plated. Thus when using PETG I limit to a lower max volumetric flow (since I mostly use the TC nozzle).
Applying your suggestion to this case (to have different extruder steps so "EM is 1") when using TC nozzle is a bit absurd, as usually extruder steps is calibrated even without a nozzle in place.

The max volumetric flow is a property of the filament to avoid filaments that are "harder to melt" to stuck and start grinding in your extruder. This limits the extrusion speed to allow it to melt. This have nothing with calibrating extruder steps.

Even if you have EM=1 for PLA, if very likely you will need a different EM using the same extruder steps for ABS. I even need slightly different EMs for different brands of ABS.

theodiem, 2023-06-08:
btw, I'm not alone there
https://ellis3dp.com/Print-Tuning-Guide/articles/misconceptions.html#you-shouldnt-tune-extrusion-multiplier-the-slicer-knows-best

CCS86, 2023-06-09:
I think you are missing my point:

Yes, different material and nozzle combinations will likely require different EM values. I am saying that those non-1 values are serving to tune the extrusion so that the _real flow rate_ actually matches the _calculated flow rate_, and therefore should **not** alter the calculated flow rate, or they will never match.

theodiem, 2023-06-13:
> 

Re-reading your messages .. I either cannot understand the correlation of what you are trying to explain with what is my bug report .. sorry :(

My issue is only with the cap set by "max volumetric speed", nothing else .. neither how the flowrate is calculated during extrusion .. only the cap set by "max volumetric speed".

In my view, if I set max volumetric speed to 20mm^3/s .. I would expect that no extrusion in the gcode exceeds that. Either with EM set to 1 , to 0.5 or to 1.5 ...

If I understood your point (which I don't agree) you suggest that if I want to limit my volumetric speed due a physical limit of my printer, I should take the actual value that I want to limit and divide it by the filament EM to actually limit to the value I actually expected.

If I want max volumetric speed to be capped at 20, and my EM is 0.95,  I need to set "max volumetric speed" to 21 .. and if my EM is 1.2 .. I  need to set it to 16.66 ... this,  to me, is a bug. "max volumetric speed" is exactly what the text says .. it is a limit I would like to apply, I shouldn't need to divide by the EM if I want to cap the max volumetric speed to a given number.

It seems you mistakenly assumed that I'm talking about how the EM math for the extrusion amounts works, and those, for me, are fine. Just how the cap is computed, in my opinion is wrong.

thghca, 2023-06-14:
You will never agree because you are talking about different multipliers. You can insert a multiplier anywhere in the calculation and you will have a completely new kind of multiplier with a different behavior. The question is whether one more parameter would be redundant or not.

CCS86, 2023-06-14:
I'm saying that if you command a 10 mm^3/s flow rate in the slicer, but find yourself using a 1.1 EM, that you are not getting 10 mm^3/s with a 1.0 EM, you are getting ~9 mm^3/s.

So then, you bump your EM up to 1.1, and now you are getting an actual flow that matches your defined limit of 10.

If the defined flow limit got multiplied by the EM, you could never correct it: 

Defined flow limit: 10
EM: 1.1
Modified flow limit: 11
Actual Flow: 10

theodiem, 2023-06-14:
@CCS86 I would actually expect what you just described. But a simple test show otherwise.
What I have expected is that if max_volumetric_flow is set to 10, the max flow should be 10 after the flow. There are tests on exactly that in gcode 👍 https://github.com/supermerill/SuperSlicer/blob/45d0532845b63cd5cefe7de7dc4ef0e0ed7e030a/src/libslic3r/GCode.cpp#L4756
>         speed = std::min(m_config.max_volumetric_speed.value / path.mm3_per_mm, speed);
it selects the min value between the capped amount by max_volumetric_speed and the computed speed in the generated gcode line.

What I observe is that:
Defined flow limit: 10
EM: 1.1
Modified flow limit: 11
Actual *maximum* Flow: 11

This is the sole reason I made the screenshots. 
Before going this discussion further to eternity, make a simple test: 
- start slicer
- add a shape-> cube
- set your EM to 1
- limit your max_volumetric_flow to 5
- on gcode preview, note that the max volumetric speed on the gcode preview is correctly displaying 5
- dont do any other change, except change your em to 1.1
- note that max volumetric speed on gcode preview now is 5.5 , even though you still have "5" set in the preset.

I would have the same as setting max speed to 100 but then having it 110 when your EM is 1.1? if you set a max_volumetric_speed, I expected it to be respected on the output gcode. The current logic do cap the max_volumetric_flow (as you will see capped to 5.5 anyways) , just it should take into consideration for EM too.

CCS86, 2023-06-15:
Sorry, I totally misunderstood the issue you reported and only now just tested it for myself.

I totally agree. 

Extrusion multiplier in filament settings is modifying the volumetric flow limitation you define and it should not do this.

hatkyinc2, 2023-11-18:
Thanks a lot for this report, this made me understand the core of the issue and at least get a workaround as I thought the feature just didn't work at all. (using a different fork, but same same for this)

supermerill, 2023-11-22:
Hello
I understand your issue.
As both (max volumetric speed & extrusion multiplier) are filament setting, on the same tab, it doesn't mater for me if it's before or after. One can modify both instead of one.
Is it more intuitive that max volumetric speed isn't multiplied by the extrusion multiplier? The multiplier multiply everything, that would make an exception.

I've follow your link to the "you-shouldnt-tune-extrusion-multiplier-the-slicer-knows-best". Now you can change the overlap, below the 100%. So this need to change the multiplier isn't there

@CCS86 what changed between your last two comment? if to have 10mm of filament, you need 11 (1.1 em), then the extrusion speed need to be 11 to not slow down the print.

theodiem, 2023-11-23:
Hello @supermerill 
First, thanks for the reply and welcome back to superslicer :D

I find way more intuitive that "max volumetric speed" is after the extrusion multiplier. If I set my max volumetric speed to 10mm^3/s, I would understand that if I have EM 0.95 or 1.10, it will never exceed 10mm^3/s.

My logic behind it is that when I measure my hotend max flow for a given filament (using Stefan's method from CNC kitchen, by printing several samples with same extrusion at different speeds and weighting them to check the deviation from the expected amount - https://www.cnckitchen.com/blog/extrusion-system-benchmark-tool-for-fast-prints), it is not related to whatever EM I would be using. To me, fine-tuning the EM is a different procedure (also first layer and top surface flow ratios) than figuring out the max flow my extruder is capable of.

I find weird to test the max flow rate of my extruder (which usually is done in mm^3/s not caring for the slicer profile as usually you extrude a given length and either listen for skipping or weight the result. Others, send a print which you gradually increase print speed while watching your printer screen for the real output flow - like in many videos from vez3d) and I have to remember to divide it by the extrusion multiplier of the filament in order to get the value I need to enter in superslicer.

Also, on superslicer, there are two variables: `max_volumetric_speed` and `filament_max_volumetric_speed` . I'm not so familiar with the source to know if one is used differently from the other or one is just an override from the other.

supermerill, 2023-11-23:
`max_volumetric_speed` is the maximum volumetric speed the auto-speed algorithm try to get. If you don't use it, it's moot.
`filament_max_volumetric_speed ` is the setting below `extrusion_multiplier`, to limit the speed of the print with this filament.
`extrusion multiplier` allow you to change the amount of plastic the extruder output, so it match what the slicer expect (ideally, you may want to create a curve, like for the ramming settings).

note:
when I'll do a rework, there may be a volumetric limit for the nozzle, and maybe the `filament_max_volumetric_speed ` can be a % of it.

ShantanuNair, 2025-04-01:
@theodiem @supermerill https://github.com/SoftFever/OrcaSlicer/issues/7307 Same issue I ended up investigating. Have any of you given thought to how we can approach this? I was considering that, instead of changing the underlying math, we can just provide a user with two interlinked inputs, one for the max flow speed, and the other for the effective max flow.