supermerill/SuperSlicer #3870: Crash on "saving as"

AmshaSharam, 2023-06-09:
### What happened?

When saving as the programs saves de project and i get error:
04:33:24: Internal error: Calling AppConfig::save() from a worker thread!
04:33:24: Internal error: Calling AppConfig::save() from a worker thread!
and closes the program.

### Project file & How to reproduce

[hvg.zip](https://github.com/supermerill/SuperSlicer/files/11701971/hvg.zip)

AmshaSharam, 2023-08-03:
I find on Prusa foruns :
https://github.com/prusa3d/PrusaSlicer/issues/7839
The problems could be generated by the grafics on laptops with 2 grafics

supermerill, 2023-11-21:
it is not reproductible? it's random?