supermerill/SuperSlicer #3874: Multitool fan speed change before tool change.

Ytmwasd, 2023-06-14:
### What happened?

Slicer puts an M106 command before tool change command, look like this without custom tool change gcode:
```
; custom gcode: end_filament_gcode
; Filament-specific end gcode 
; custom gcode end: end_filament_gcode
M106 S48.45
T1
G92 E0
; custom gcode: start_filament_gcode
; Filament gcode
; custom gcode end: start_filament_gcode
```
with custom toolchange gcode:
```
; custom gcode: toolchange_gcode
M106 S48.45 <- at this point printer doesn`t know about changing extruder, so it's apply this to the current extruder 
T1 <- and now the extruder has changed. Too late the fan is blowing not on the extruder that prints.
; custom gcode end: toolchange_gcode
```
In this case, the printer always blows at maximum on ABS, and does not blow at all on PLA
T1/E0 must be before any changes, otherwise all changes are applied to the extruder that has finished its work.
or M106 commant shold look like this: `M106 P<fan> S<speed>`

In "custom tool for changing gcode" I could write `M106 P{next_extruder} S{<some variable for the fan speed>}`, but there is no variable for that, as its generated later.

### Project file & How to reproduce

Here an example gcode and project:
[multitool.zip](https://github.com/supermerill/SuperSlicer/files/11748884/multitool.zip)

### Version

2.4.58.5

### Operating system

windows 11

### Printer model

custom, klipper, idex

Ytmwasd, 2023-06-14:
Same here: https://github.com/supermerill/SuperSlicer/discussions/1253

legend069, 2023-10-21:
can you check if this bug is resolved with my PR ?
https://github.com/supermerill/SuperSlicer/pull/3735

supermerill, 2023-11-22:
The fan change is trigger after the toochange, but also when it encounter a 'T' command.
I removed the second trigger, as i don't see where that can be useful.

closed 2024-02-11 by supermerill

ppisl, 2025-03-19:
I believe this issue is not yet fully resolved, or there has been a regression. I'm using Linux, and in PrusaSlicer 2.9.1, M106 Sxxx is still being generated before the Tx tool change command. My Tool change G-code is as follows:

;Tool change G-code
{if wipe_tower}
G91
G1 Z1 F7800
G90
G1 X{wipe_tower_x} Y{wipe_tower_y} F{travel_speed*60}
G91
G1 Z-1 F7800
G90
{endif}
M106 P{current_extruder} S0 ; disable fan part
{if (layer_num + 1) > disable_fan_first_layers{next_extruder} }
M106 P{next_extruder} S{min_fan_speed{next_extruder} * 2.55}
{endif}
; next line with M106 should be removed

However, the generated G-code still includes lines like:

M106 S255
T1

This indicates that the fan is being set on the previous tool (T0 in this case) because it is still the active tool at that point. I've worked around this by adding a G-code substitution in the Print Settings, which replaces:

; next line with M106 should be removed\nM106 S

with:

; the next line was commented out by gcode substitutions\n; M106 S

The resulting G-code can be seen in the attached picture.

[image]

I don't have rights to reopen this issue. Should I create new one?