supermerill/SuperSlicer #3875: Improper first layer of large print

hansratzinger, 2023-06-17:
### Version
Version 2.4.58.5

### Operating system type + version
Win 10 Version 22H2 Build 19045.3086

### Behavior
* Slicing a very large print (185 x 226 x 235 mm) for Anycubic Vyper (max 245 x 245 x 260).
[image]

After slicing the first layer represents only a fraction of the entire bottom of the model.
[image]

* Steps:
1. import model
KT-Muffe_DN150-20_Grad_mit_50_Schlauchanschluss_teilbar v18.3mf
2. rotate x 90°
3. place on face
4. slice now
5. go to level 1 bottom view 
[image]

* Result in Cura:
[image]

#### Project File (.3MF) where problem occurs
[KT-Muffe_DN150-20_Grad_mit_50_Schlauchanschluss_teilbar v18.zip](https://github.com/supermerill/SuperSlicer/files/11778155/KT-Muffe_DN150-20_Grad_mit_50_Schlauchanschluss_teilbar.v18.zip)

Arakon, 2023-06-17:
Your model isn't 90° flat on that side. It's at a slight angle.
Using 2.5.59.2, after "place on face", it looks fine.
[image]

hansratzinger, 2023-06-17:
The same bug appears in PursaSlicer.
https://github.com/prusa3d/PrusaSlicer/issues/10831

hansratzinger, 2023-06-18:
Many thanks @Arakon . I used F (Place on face) but did not know to klick afterwards at the bottom. Now it is working!
Same issue in https://github.com/prusa3d/PrusaSlicer/issues/10831#issuecomment-1595693686

closed 2023-06-18 by hansratzinger