supermerill/SuperSlicer #3877: Can not install in OSX

AnakinSpain, 2023-06-19:
### What happened?

trying to install

### Project file & How to reproduce

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [77496]
Path:                  /Volumes/VOLUME/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-06-19 21:07:02.3552 +0200
OS Version:            macOS 13.4 (22F66)
Report Version:        12
Anonymous UUID:        8FA44F8E-4E22-0588-01C4-F260FB6CFC76

Sleep/Wake UUID:       9E3D3296-81FF-4687-8D76-D6320D6ED7A4

Time Awake Since Boot: 1200000 seconds
Time Since Wake:       1928 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: <EA3F7209-F390-3268-9C90-D8081DCEDC30> /Volumes/VOLUME/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file, not in dyld cache)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   Rosetta Runtime Routines	       0x1074539a8 ???
1   <translation info unavailable>	       0x1074d5954 ???
2   dyld                          	       0x203deffd7 abort_with_payload_wrapper_internal + 82
3   dyld                          	       0x203df0009 abort_with_payload + 9
4   dyld                          	       0x203d748f0 dyld4::halt(char const*) + 375
5   dyld                          	       0x203d71b71 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 4526
6   dyld                          	       0x203d703bd start + 1805

Thread 1:: com.apple.rosetta.exceptionserver
0   runtime                       	    0x7ff7ffc0f694 0x7ff7ffc0b000 + 18068

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000006   x1: 0x0000000000000001   x2: 0x0000000308658870   x3: 0x00000000000000d8
    x4: 0x0000000308658470   x5: 0x0000000000000000   x6: 0x0000000000000000   x7: 0x0000002080d85d88
    x8: 0x0000000308658470   x9: 0x0000000000000000  x10: 0x00000000000000d8  x11: 0x0000000308658470
   x12: 0x00000000000000d8  x13: 0x0000000308658870  x14: 0x0000000000000001  x15: 0x0000000000000006
   x16: 0x0000000000000209  x17: 0x0000000000000000  x18: 0x800000010f4b8b80  x19: 0x0000000107449000
   x20: 0x00000000000007ff  x21: 0x000000010f4bdfa0  x22: 0xffffffffffffffff  x23: 0x000000010745397c
   x24: 0x0000000203deffd7  x25: 0x0000000107453248  x26: 0x00000001074d3ba8  x27: 0x0000000000000000
   x28: 0x0000000203dd5c42   fp: 0x000000002c000000   lr: 0x00000001074d5954
    sp: 0x000000010f4b8b80   pc: 0x00000001074539a8 cpsr: 0x80001000
   far: 0x0000000203deff85  esr: 0x56000080  Address size fault

Binary Images:
       0x203d6a000 -        0x203e05fff dyld (*) <9e98a840-a3ac-31c1-ab97-829af9bd6864> /usr/lib/dyld
    0x7ff7ffc0b000 -     0x7ff7ffc3afff runtime (*) <c4d322c3-30c5-3c52-adb8-2fd12ec92bc0> /usr/libexec/rosetta/runtime
       0x10f453000 -        0x10f4a6fff libRosettaRuntime (*) <c8f25cdc-b334-3f67-bc30-50e9852f8da2> /Library/Apple/*/libRosettaRuntime
       0x1007f0000 -        0x1028c3fff com.merill.slic3r/ (SuperSlicer SuperSlicer_2.4.58.5_2022-09-23) <ea3f7209-f390-3268-9c90-d8081dcedc30> /Volumes/VOLUME/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
[... 177 lines cut ...]
      "deploymentId" : 900000005
    }
  ]
}
}

Model: iMac21,2, BootROM 8422.121.1, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: iMac, Retina 4.5K (4480 x 2520), Main, MirrorOff, Online
Memory Module: LPDDR4, Micron
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: Jan 12 2023 05:51:25 version 18.20.383.14.7.8.149 FWID 01-f03416d4
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: iMac, Apple Inc.
Thunderbolt Bus: iMac, Apple Inc.

### Operating system

Mac OS ventura 13.4

### Printer model

ender 3

mtaumike, 2023-06-20:
This has been an issue for a while and you have 2 choices:
- build yourself from source and it will run
- use the Intel version and allow Rosetta2 to do things; I don't find this slow

mtaumike, 2023-06-23:
> Use 2.5.59.0 https://github.com/supermerill/SuperSlicer/releases/tag/2.5.59.0

This release works if the user uses the Intel build as the ARM build (M-series) still fails anyway.

AnakinSpain, 2023-06-24:
I have one  M1

supermerill, 2023-11-21:
see supermerill/SuperSlicer#3948
arm will comme, as now there seems to be arm runner on github.

closed 2023-11-21 by supermerill