supermerill/SuperSlicer #3884: Following PS 2.6.0

LazaroFilm, 2023-06-28:
Now that 2.6.0 is out of beta on PrusaSlicer, so we have a timeframe for SuperSlicer to follow suit? Prusa is sadly missing some tools that I became dépendant on in SS, namely the single perimeter on top and bottom layers.

CCS86, 2023-06-29:
Don't hold your breath. They get further apart every day.

freighttt, 2023-07-10:
> PrusaSlicer gets updated very often with very cool features, but they make the baffling decision to not include basic settings, like single perims on top/bottom layers like you mentioned.
> 
> Honestly, I'd look into OrcaSlicer at this point. All the big features from PrusaSlicer 2.6 have already been ported over to OrcaSlicer. Orca also lets you use single perimeters for top and bottom layers.
> 
> I consider SuperSlicer an end of life product, at the moment. Don't count on literally any meaningful updates. The last action was over 2 months ago, and those actions were for minor things. If you're hoping for SuperSlicer to get PrusaSlicer 2.6 features you're going to be waiting for a very long time, if ever.

its definitely not end of life. its a one man show and he had some life stuff happening. nightly still gets updated fairly regularly and merill is still around on discord. just be patient and toss him a few bucks

jeremytodd1, 2023-07-10:
The last actions for SuperSlicer were made 2 days ago it looks like. Before that, the last actions were done over 3 months ago. That is far from regular updates. 

For comparison, in the last **4 days** there has been 28 actions on OrcaSlicer. To get to 28 actions on SuperSlicer we'd need to go back literally 5 months. I know this isn't a great comparison as actions can be tiny changes, I just wanted to throw this out there as I think it'd a good indicator.

I'm all for financially supporting development work on open source software like this. I think it's great. I also want my money to go to a project where I think it might actually matter. I'm not going to throw money at someone who hasn't shown much progress in the last year. SS is still missing features from PrusaSlicer 2.5, which is quite old at this point.

I'm grateful for all the work that supermerill has put into SS. I used the software exclusively for a few thousand print hours. If it's at the point where supermerill either isn't motivated to work on it anymore, or simply doesn't have the time due to life happenings, I believe that it'd be a good idea for someone to fork SuperSlicer so it can have more consistent development.

There is a reason that more and more people are looking into other slicers. Slicers are moving very quickly at the moment with some huge features. They don't want to miss out on those and I don't blame them.

Weskus0203, 2023-07-16:
SuperSlicer does seem dead. Gut wrenching since it was by far the best slicer I have used. I could never replicate the print quality on Cura or PS. Thinking of giving Orca a try but I still have hope that SS will return one day.

LAP87, 2023-07-26:
Orca is also missing a bunch of the deep-dive settings i'm used to tinkering with, i'm sticking with nightly releases of SS until it's truly abandoned.

supermerill, 2023-11-22:
I'll go to 2.7 directly

vkhodygo, 2023-11-24:
@supermerill Any ETA? What's really missing is proper organic supports and arc overhangs.