supermerill/SuperSlicer #3889: Crash when trying to slice more than 2 instances of an object with "complete individual objects" turned on

daffy1234, 2023-07-05:
### What happened?

I added a model for a D6 dice, turned on "complete individual objects", set it to 5 instances, and told it to arrange them. It did, but when I went to slice it, it froze and then crashed.

Below is a log of the terminal output of superslicer

```
04:08:20 PM: Debug: Adding duplicate image handler for 'JPEG file'
[2023-07-05 16:08:21.384621] [0x00007fa7bffff6c0] [error]   Unable to open connection to spacenavd
04:08:21 PM: Debug: Unrecognized accel key 'Dèl', accel string ignored.
04:08:21 PM: Debug: "Shift+Tab" is not supported as a keyboard accelerator with GTK
04:08:22 PM: Debug: window wxTreeCtrl(0x55ab99528210) lost focus even though it didn't have it
04:08:22 PM: Debug: window wxTreeCtrl(0x55ab99528210) lost focus even though it didn't have it
04:08:22 PM: Debug: window wxTreeCtrl(0x55ab99528210) lost focus even though it didn't have it
04:08:22 PM: Debug: window wxTreeCtrl(0x55ab99528210) lost focus even though it didn't have it
04:08:22 PM: Debug: window wxTreeCtrl(0x55ab99528210) lost focus even though it didn't have it
04:08:28 PM: Debug: window wxTreeCtrl(0x55ab99528210) lost focus even though it didn't have it
04:08:28 PM: Debug: window wxTreeCtrl(0x55ab99528210) lost focus even though it didn't have it
04:08:28 PM: Debug: window wxTreeCtrl(0x55ab99528210) lost focus even though it didn't have it
04:08:28 PM: Debug: window wxTreeCtrl(0x55ab99528210) lost focus even though it didn't have it
04:08:28 PM: Debug: window wxTreeCtrl(0x55ab99528210) lost focus even though it didn't have it
/usr/include/c++/13.1.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = double; _Alloc = std::allocator<double>; reference = double&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)
```

### Project file & How to reproduce

Opening this project file and slicing it causes a crash every time for me
[crash.zip](https://github.com/supermerill/SuperSlicer/files/11961539/crash.zip)

### Version

2.4.58.5

### Operating system

Arch Linux

### Printer model

Creality Ender 5 s1

zark0-UwU, 2023-11-07:
Same problem here!

supermerill, 2023-11-21:
found, fixed

closed 2024-02-11 by supermerill