supermerill/SuperSlicer #3890: Increasing Travel Speed and Accel INcreases Print Time

CCS86, 2023-07-09:
### What happened?

Anyone know what the hell is going on here?

Changing only the travel speed and accel to significantly higher values is increasing the calculated print time by 18 minutes.

https://github.com/supermerill/SuperSlicer/assets/12439543/784710c8-a4b0-4f66-84d6-b6ffc3af950e

### Project file & How to reproduce

...

### Version

2.5.59.2

### Operating system

Windows 10 x64

legend069, 2023-07-29:
i'm going to assume the speed increase has messed with your min layer time variable

CCS86, 2023-07-29:
> i'm going to assume the speed increase has messed with your min layer time variable

That's not possible. Even if I was already clipped to minimum layer time, any increases in speed or acceleration would be compensated for, giving a net 0 change in layer time.

The reality is that my layer times are like 40+ seconds and my layer time goal is set to 4 seconds.

legend069, 2023-07-29:
> That's not possible. Even if I was already clipped to minimum layer time, any increases in speed or acceleration would be compensated for, giving a net 0 change in layer time.
>
> The reality is that [...]

Can you please slice both versions and upload here.

CCS86, 2023-07-29:
Here is a test file.

Just slice, then bump travel speed from 170 up to 350 and slice again. Forget accel. That seems to work correctly. 

[Travel-speed-test.3mf.txt](https://github.com/supermerill/SuperSlicer/files/12208133/Travel-speed-test.3mf.txt)

legend069, 2023-08-07:
thanks for that. it looks like it's an issue with "reprapfirmware" this has narrowed down the search slightly
due to the large exported.gcode file i'm unable to investigate any further to see what functions are bugged.

after changing the g-code flavor from reprapfirmware to klipper
the estimated time has changed
changed speed to 170 with config changed to reprapfirmware 
[image]

changed speed to 350 with config changed to reprapfirmware 
[image]

changed speed to 170 with config changed to klipper
[image]

changed speed to 350 with config changed to klipper
[image]

CCS86, 2023-08-07:
Interesting. So, it's most likely that the lower speed has an unrealistically low time estimate, and the higher speed is more accurate.

supermerill, 2023-11-22:
the time estimation is based on marlin.
Maybe there's something in reprap flavor that mess with it?
The only things that may cause that is the difference in the M203 M204 M205 M566 .

Nerobeus, 2024-02-11:
[part.zip](https://github.com/supermerill/SuperSlicer/files/14231757/part.zip)
sorry for mistake, here is the project

Nerobeus, 2024-02-11:
[part.zip](https://github.com/supermerill/SuperSlicer/files/14232542/part.zip)