supermerill/SuperSlicer #3894: SuperSlicer Crashes when turning on Spiral Vase mode

ibkozi, 2023-07-16:
### Version
[_Version of SuperSlicer used goes here (help->about)_]
Version: 2.5.59.2
Build: SuperSlicer_2.5.59.2
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 22621), 64-bit edition
Total RAM size [MB]: 34,270MB

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 22621), 64-bit edition

### Behavior
Program crashes after enabling Spiral Vase Mode, then clicking OK to change the rest of the settings that need to change as well.
* _Steps needed to reproduce the problem_
Turn on SuperSlicer, enable Spiral Vase Mode, Click OK when it asks to change the rest of the settings for you.
* _Expected Results_
Not to crash.  Enter Spiral Vase Mode

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

Pravv, 2023-07-16:
It's already fixed in the latest nightly release. You can find it under https://github.com/supermerill/SuperSlicer/actions

VanessaE, 2023-07-20:
If you'd rather avoid the nightly builds, use release 2.5.59.0.  Spiral vase mode works fine there.

closed 2024-02-11 by supermerill