supermerill/SuperSlicer #3896: Crash on slice

flockman66, 2023-07-18:
### What happened?

When attempting to slice the main body for this model: https://www.thingiverse.com/thing:6086143 

on superslicer the program crashes without warning, after opening the program from the terminal i see the following output: 

superslicer: /usr/src/debug/superslicer/SuperSlicer/src/libslic3r/ExtrusionEntity.hpp:461: virtual const Slic3r::Point& Slic3r::ExtrusionLoop::last_point() const: Assertion `this->first_point() == this->paths.back().polyline.points.back()' failed.

zsh: IOT instruction (core dumped)  superslicer

### Project file & How to reproduce

1. download file
2. open BE-body_.stl in superslicer
3. slice with the default voron 2.4 350 slicer 0.4mm nozzle at 0.2mm  profile for abs
4. crash should follow

### Version

2.4.58.5-6

### Operating system

Manjaro linux - Gnome

### Printer model

voron 2.4 with stealthburner

Creo2005, 2023-07-18:
2.4.58.5-6 - very old version :)

flockman66, 2023-07-18:
ah sorry, i though manjaro was a rolling release so i assumed its the latest, too bad theres no flatpak yet :P

flockman66, 2023-07-18:
i see theres an appimage so i'll give that a go

flockman66, 2023-07-18:
Yeah can confirm it works with 2.5.59, sorry for taking up the time :)

closed 2023-07-18 by flockman66