supermerill/SuperSlicer #3903: Crash when sinking, or cutting object.

Chacha75, 2023-07-26:
### What happened?

I wanted to print the upper part of a model, but SuperSlicer crashes when I try to cut it, or trying to sink under the bed.

### Project file & How to reproduce

[Corner panel clip 4mm.zip](https://github.com/supermerill/SuperSlicer/files/12176189/Corner.panel.clip.4mm.zip)

Open the attached project, on the right pane, in Object manipulation edit the position Z coordinate to 0, and hit enter. 

### Version

2.4.58.5

### Operating system

Manjaro 23.0.0

### Printer model

Custom

Creo2005, 2023-07-27:
2.4.58.5 - very old!! 
use 2.5.59.2 night build! or 2.5.59.0 relise

Chacha75, 2023-07-28:
2.5.59.2 works perfectly!
Thank you!

closed 2023-07-28 by Chacha75