supermerill/SuperSlicer #3905: Vasemode crashes Superslicer

VHelst, 2023-07-28:
### Version
_Version of SuperSlicer used goes here (help->about)_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

VHelst, 2023-07-28:
Hi!
When I tick the "spiral vase" box in Superslicer it crashes and closes everytime...
I have a Sovol SV05, Windows 10 and I´ve tried with many different files of vases, but the same thing happens eveytime.
V

Creo2005, 2023-07-30:
> Hi! When I tick the "spiral vase" box in Superslicer it crashes and closes everytime... I have a Sovol SV05, Windows 10 and I´ve tried with many different files of vases, but the same thing happens eveytime. V

use night buil 2.5.59.2 - all is ok. https://github.com/supermerill/SuperSlicer/actions/runs/5548343816

VHelst, 2023-07-30:
Unfortunately, that´s the one I am using. Maybe I´m suffering from some bug or something...
I think the slicer works absolutely perfekt for everything else, and it´s the best one I´ve tried.
It´s just to bad my copy seems to hate vasemode :-).
V

Hun73rdk, 2023-08-04:
The newest version of SuperSlicer crashes if you turn on vase mode i can confirm that i went back to build before that version 2.4.58.5 to get rid of the crashing

EvilGremlin, 2023-08-23:
Fixed already, download nidhtly-dev branch build from actions.

Pelado-Mat, 2023-10-26:
> Fixed already, download nidhtly-dev branch build from actions.

The artifact has expired, could we get another run to get the binaries?
[image]

Kadah, 2023-11-11:
> > Fixed already, download nidhtly-dev branch build from actions.
> 
> The artifact has expired, could we get another run to get the binaries? [image]

@Pelado-Mat I forked and triggers a build last night. I have not tested it yet as the build was going to take almost 1.5 hours so I end up just using an older version instead for the current print project I needed to run.

https://github.com/Kadah/SuperSlicer/actions/runs/6822230742

supermerill, 2023-11-21:
will create a new release soon,  just checking if i'm not missing a serious bug.

closed 2024-02-11 by supermerill