supermerill/SuperSlicer #3908: Seam Position: 'Corners' and 'Nearest' options not selectable

Packman65, 2023-07-31:
### What happened?

I'm using Superslicer 2.5.59.2. In the Seam Position setting, the top two options, "Corners" and "Nearest" are not selectable from the pull-down list. If any of these are selected, it ends up being "Custom". Is this normal? Am I required to do something else to make either of these two settings sticky?
I've attached a video to illustrate what I mean.

Still on Seam Position, which option emulates Cura's "Sharpest" and "Seam hiding" seam position feature? Would that be "Corners"?

Thanks.

### Project file & How to reproduce

[video_20230731_185713.zip](https://github.com/supermerill/SuperSlicer/files/12216450/video_20230731_185713.zip)

### Version

2.5.59.2

### Operating system

Win 10 Enterprise 22H2

### Printer model

SeeMeCNC Rostock Max v3

gsokoll, 2023-08-31:
Same issue here.  v2.5.59.2 on Win11

supermerill, 2023-11-27:
corner are nearest are in fact just custom with pre-filled weights.
If the weight are exactly the ones that "corner" use, it should be selected.
I'll check if the script isn't acting weird.