supermerill/SuperSlicer #3912: SuperSlicer demands old OpenCASCADE

l29ah, 2023-08-03:
`dev` branch wants version 7.6.2 and isn't satisfied by 7.6.3 for some reason, even though those should be compatible. Gentoo only ships 7.6.3 and 7.7.0, so it's awkward to build SuperSlicer there.

RealDeuce, 2023-08-11:
I had the same issue on FreeBSD (where I use 7.7.0), and just couldn't get the version range to work and had to locally patch.

Assuming Gentoo has an ebuild, patching `src/occt_wrapper/CMakeLists.txt` works for me.

RealDeuce, 2023-08-11:
To add some details, when I change the find_package to `find_package(OpenCASCADE 7.6.2...7.99.99 REQUIRED)`, I get:

```
CMake Warning (dev) at /usr/local/lib/cmake/OpenCASCADEConfigVersion.cmake:13 (message):
  `find_package()` specify a version range but the version strategy
  (ExactVersion) of the module `OpenCASCADE` is incompatible with this
  request.  Only the lower endpoint of the range will be used.
Call Stack (most recent call first):
  src/occt_wrapper/CMakeLists.txt:22 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at src/occt_wrapper/CMakeLists.txt:22 (find_package):
  Could not find a configuration file for package "OpenCASCADE" that is
  compatible with requested version range "7.6.2...7.99.99".

  The following configuration files were considered but not accepted:

    /usr/local/lib/cmake/OpenCASCADEConfig.cmake, version: 7.7.0
```

So this looks like a CMake issue... for some reason it's assuming EXACT even though it's not specified.

RealDeuce, 2023-08-11:
Digging in further, this is caused by OpenCASCADE... from their OpenCASCADEConfigVersion.cmake file:

```
if (PACKAGE_FIND_VERSION_RANGE)
  message(AUTHOR_WARNING
    "`find_package()` specify a version range but the version strategy "
    "(ExactVersion) of the module `${PACKAGE_FIND_NAME}` is incompatible "
    "with this request. Only the lower endpoint of the range will be used.")
endif()
```

So SS will need to either not specify a version at all and not be able to inform the user of the problem, or SS will need to implement version checking itself in the cmake file.

RealDeuce, 2023-08-11:
It looks like:

```
find_package(OpenCASCADE REQUIRED)
if (OpenCASCADE_VERSION VERSION_LESS 7.6.2)
    message(FATAL_ERROR "OpenCASCADE v7 >= v7.6.2 must be used")
elseif (OpenCASCADE_VERSION VERSION_GREATER_EQUAL 8.0.0)
    message(FATAL_ERROR "OpenCASCADE v7 >= v7.6.2 must be used")
endif ()
```

Will work.

supermerill, 2023-11-27:
If it's still aproblem in the 2.7, i'll add it.

kc2zgu, 2024-12-16:
Looks like there's some incompatibility with OpenCASCADE 7.8, which is the only version Gentoo has now. Some libs like libTKSTEP have been renamed or moved around according to https://dev.opencascade.org/doc/overview/html/occt__upgrade.html#upgrade_780_tk_rework

RealDeuce, 2024-12-29:
Yeah, I did this on my system:
```
diff --git a/src/occt_wrapper/CMakeLists.txt b/src/occt_wrapper/CMakeLists.txt
index 55b03120f1..503653b5d2 100644
--- a/src/occt_wrapper/CMakeLists.txt
+++ b/src/occt_wrapper/CMakeLists.txt
@@ -27,11 +27,6 @@ elseif (OpenCASCADE_VERSION VERSION_GREATER_EQUAL 8.0.0)
 endif ()
 
 set(OCCT_LIBS
-    TKXDESTEP
-    TKSTEP
-    TKSTEP209
-    TKSTEPAttr
-    TKSTEPBase
     TKXCAF
     TKXSBase
     TKVCAF
@@ -55,6 +50,20 @@ set(OCCT_LIBS
     TKernel
 )
 
+if (OpenCASCADE_VERSION VERSION_LESS 7.8.0)
+    list(APPEND OCCT_LIBS
+        TKXDESTEP
+        TKSTEP
+        TKSTEP209
+        TKSTEPAttr
+        TKSTEPBase
+    )
+else()
+    list(APPEND OCCT_LIBS
+        TKDESTEP
+        TKXSDRAWSTEP
+    )
+endif()
 target_include_directories(OCCTWrapper PRIVATE ${CMAKE_CURRENT_BINARY_DIR})
 target_include_directories(OCCTWrapper PUBLIC ${OpenCASCADE_INCLUDE_DIR})
 target_link_directories(OCCTWrapper PUBLIC ${OpenCASCADE_LIBRARY_DIR})
```

kc2zgu, 2025-03-31:
Another thing I noticed trying to get the latest version working on Gentoo related to this: Both PrusaSlicer and SuperSlicer install the same /usr/bin/OCCTWrapper.so file, so they can't be installed on the same system. Probably makes sense to open a separate issue, but I think SuperSlicer should rename its copy to avoid a collision.

msrd0, 2025-04-01:
@kc2zgu feel free to steal my patch: https://aur.archlinux.org/cgit/aur.git/tree/0010-occtwrapper-install-dir.patch?h=superslicer-prerelease

supermerill, 2025-04-01:
> [@kc2zgu](https://github.com/kc2zgu) feel free to steal my patch: https://aur.archlinux.org/cgit/aur.git/tree/0010-occtwrapper-install-dir.patch?h=superslicer-prerelease

patch added to upstream.

note: the OpenCASCADE version issue is resolved in 2.7, right?

msrd0, 2025-04-01:
Nope, on 2.7.61.0 I still get this error message without my patches:

```
CMake Error at src/occt_wrapper/CMakeLists.txt:22 (find_package):
  Could not find a configuration file for package "OpenCASCADE" that is
  compatible with requested version "7.6.2".

  The following configuration files were considered but not accepted:

    /usr/lib/cmake/opencascade/OpenCASCADEConfig.cmake, version: 7.8.1
    /lib/cmake/opencascade/OpenCASCADEConfig.cmake, version: 7.8.1
```

supermerill, 2025-04-01:
problem is, I can't update your patches.

One is: 
```
-find_package(OpenCASCADE 7.6.2 REQUIRED)
+find_package(OpenCASCADE REQUIRED)
+if (OpenCASCADE_VERSION VERSION_LESS 7.6.2)
+    message(FATAL_ERROR "OpenCASCADE v7 >= v7.6.2 must be used")
+elseif (OpenCASCADE_VERSION VERSION_GREATER_EQUAL 8.0.0)
+    message(FATAL_ERROR "OpenCASCADE v7 >= v7.6.2 must be used")
+endif ()
```

But i have : `find_package(OpenCASCADE REQUIRED)`

Do you have patches updated for 2.7?

supermerill, 2025-04-02:
oh, sorry. I may have merged a pull request that changes it in the meantime.