supermerill/SuperSlicer #3913: Missing some retractation in single extruder MM

elpopo-eng, 2023-08-04:
### What happened?

Some retractions are missed when moving to the next fill area, so the color is contaminated by other colors along the way.
In layer 3 is more visible
[image]

### Project file & How to reproduce

[BugSusie_Milti.zip](https://github.com/supermerill/SuperSlicer/files/12261023/BugSusie_Milti.zip)

### Version

2.5.59.2

### Operating system

Windows 10

### Printer model

ERCF

bchnrq, 2023-08-26:
Hi, i'm experiencing the same issue :(
miss some Z-hop for travels. beetween pieces.

bchnrq, 2023-08-26:
i think i found something 
uncheck in print settings > Infill > Advanced > Only retract when crossing perimeters

closed 2024-02-29 by elpopo-eng