supermerill/SuperSlicer #3914: Superslicer 2.4.58.5 crashes on opening

drpeej, 2023-08-07:
### What happened?

I have installed this on my ARM Macbook but when I try it on my Intel Mac it crashes without opening.

This is the message I get

Can someone advise please

Peter

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [3392]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-08-07 11:54:54.3935 +0100
OS Version:            macOS 13.4.1 (22F770820d)
Report Version:        12
Bridge OS Version:     7.5 (20P5060)
Anonymous UUID:        BEDB91D9-2C16-AA80-A47A-77F741B80B2B

Time Awake Since Boot: 4200 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: <EA3F7209-F390-3268-9C90-D8081DCEDC30> /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file, not in dyld cache)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	    0x7ff80b4bfc42 __abort_with_payload + 10
1   dyld                          	    0x7ff80b4d9fd7 abort_with_payload_wrapper_internal + 82
2   dyld                          	    0x7ff80b4da009 abort_with_payload + 9
3   dyld                          	    0x7ff80b45e8f0 dyld4::halt(char const*) + 375
4   dyld                          	    0x7ff80b45bb71 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 4526
5   dyld                          	    0x7ff80b45a3bd start + 1805

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7b36f5448  rdx: 0x00007ff7b36f58b0
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7b36f5490  rsp: 0x00007ff7b36f5448
   r8: 0x00007ff7b36f54b0   r9: 0x0000000000000000  r10: 0x00000000000000d8  r11: 0x0000000000000246
  r12: 0x00000000000000d8  r13: 0x00007ff7b36f58b0  r14: 0x0000000000000001  r15: 0x0000000000000006
  rip: 0x00007ff80b4bfc42  rfl: 0x0000000000000246  cr2: 0x000000010fa4b000
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x10c809000 -        0x10e8dcfff com.merill.slic3r/ (SuperSlicer SuperSlicer_2.4.58.5_2022-09-23) <ea3f7209-f390-3268-9c90-d8081dcedc30> /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
    0x7ff80b454000 -     0x7ff80b4ec5cf dyld (*) <5db85b72-c63a-3182-91e5-5c942ec30e48> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
[... 133 lines cut ...]
    },
    {
      "treatmentId" : "a092db1b-c401-44fa-9c54-518b7d69ca61",
      "experimentId" : "64a844035c85000c0f42398a",
      "deploymentId" : 400000016
    }
  ]
}
}

### Project file & How to reproduce

Download installer and run it

### Version

2.4.58.5

### Operating system

Macos Ventura

### Printer model

Voron 2.4

Creo2005, 2023-08-07:
Work fine for mr in MacOS Ventura - night build https://github.com/supermerill/SuperSlicer/actions/runs/5548343814  nightly_macos.dmg

supermerill, 2023-11-21:
i'll look at the mac build again after the merge.