supermerill/SuperSlicer #3916: Thumbnails Not Showing

drizztguen77, 2023-08-09:
### What happened?

Thumbnails normally work just fine. But this one specific file will not display them for some reason. It also doesn't display the normal stats (filament and print time). You can see it in the attached screenshot. I sliced the same object in Prusaslicer and it shows the thumbnails just fine. I tried in both the latest release and nightly release of SuperSlicer and they don't show in either one. I have tried to compare the 2 gcode files the best I could and could not find anything different that might cause this problem. I even tried copying the thumbnail data out of the working gcode file into the non working file and it still did not display the thumbnails. So I'm not sure what is causing it. I've attached both the working and non-working gcode files in the project zip file. There are 3 gcode files and both SuperSlicer gcode files do not display thumbnails. The Prusa gcode file does display them. All my other .stl files work fine so I suspect it is an issue with the .stl file but I'm not certain.

[image]

### Project file & How to reproduce

Drop the attached .stl into a new project in SuperSlicer and then slice it, save the gcode and then open the gcode file in Mainsail.
[files.zip](https://github.com/supermerill/SuperSlicer/files/12298473/files.zip)

### Version

2.5.59.2 and Nightly

### Operating system

windows 10

### Printer model

voron 2.4 with Stealthburner

donduce, 2023-08-29:
It's the name of your stl, turn the ü into a ue and the problem should be fixed. I've also been looking for the wolf why I had no thumbnails

drizztguen77, 2023-08-29:
Yep, that works. I think the bug still should be fixed in the code but changing the ü into a standard English letter did resolve the issue.

closed 2023-11-27 by supermerill